TheSpaghettiDetective/obico-server #156: Detective not watching

krackware, 2020-01-20:
[image]

I just installed The Spaghetti Detective on my Ubuntu server.  The connection to octoprint works fine and the majority of the setup was uneventful.  

However, when the software runs, it warns that the detective is not watching (see attached).  It appears that it may be asking for me to purchase time (hourglass icon top right).  But there is no way to do this for a home server.  Any ideas?

RaymondHimle, 2020-01-20:
Best place to get responsive help for TSD is on the TSD Discord https://discord.gg/hsMwGpD

RaymondHimle, 2020-02-03:
@kennethjiang You can close this due to inactivity

closed 2020-04-18 by kennethjiang