TheSpaghettiDetective/obico-server #157: Cant send e-mails due to e-mail verification not working

ITredneck, 2020-01-21:
I am unable to send e-mails due to not being able to verify my account. There is a bug requiring the verification before it can send.

Quote from @kennethjiang 
just checked the code. It looks like the intention was to skip email verification for private servers as the default behavior, which makes sense
but somehow it requires verification anyway, which looks like a bug
so I think it's fair to open an issue for it

joachimvenaas, 2020-02-27:
Just fixed this problem myself, resend verification mail (if you’re not getting it, check your smtp settings). if you are getting the verification mail and the link is not working, replace https with http

kennethjiang, 2020-04-18:
Closing due to inactivity.

closed 2020-04-18 by kennethjiang