TheSpaghettiDetective/obico-server #160: Compose failed "ml_api failed to build"

IIPoliII, 2020-01-27:
Hey i am trying to compose this container but i can't make it to work.

I saw a similar issue : https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/issues/35 but in my case i can't use this old version of docker on my ubuntu 18.04.

```bash
 2020-01-27 14:29:25 ⌚  vlpi01 in ~/TheSpaghettiDetective
± |master ✓| → sudo docker-compose up -d
Building ml_api
Step 1/8 : FROM thespaghettidetective/ml_api:base
 ---> 6eaa0374427c
Step 2/8 : WORKDIR /app
 ---> Using cache
 ---> 5ec694df9cff
Step 3/8 : EXPOSE 3333
 ---> Using cache
 ---> f4aab7e39c32
Step 4/8 : ADD . /app
 ---> Using cache
 ---> 404a0d2106ea
Step 5/8 : RUN cp bin/model_$(uname --m).so bin/model.so &&     cp bin/model_gpu_$(uname --m).so bin/model_gpu.so
 ---> Using cache
 ---> 76e36e03d07a
Step 6/8 : RUN pip install --upgrade pip
 ---> Using cache
 ---> ea16762d92c7
Step 7/8 : RUN pip install -r requirements.txt
 ---> Using cache
 ---> 711d4ac3848f
Step 8/8 : RUN wget --quiet -O model/model.weights $(cat model/model.weights.url | tr -d '\r')
 ---> Running in 9197d3c26dbe
ERROR: Service 'ml_api' failed to build: The command '/bin/sh -c wget --quiet -O model/model.weights $(cat model/model.weights.url | tr -d '\r')' returned a non-zero code: 4

```

RaymondHimle, 2020-01-27:
Try running "sudo docker-compose --verbose up -d

IIPoliII, 2020-01-27:
Thanks for your reply, i founded the issue just now. After a lot of testing i was a bit stupid at the end. 
I am using Proxmox and i was trying to make it in a Container using the freature nesting from proxmox normally it works great but not with all containers. And creating a container in a container is a bit weird too.

So basicly i think creating a vm is the solution ! Now it works perfectly C:

closed 2020-01-27 by IIPoliII