TheSpaghettiDetective/obico-server #162: Qnap NAS plugin

simonduz, 2020-02-02:
An enhancement request to publish a Qnap NAS plugin for this install.    Qnap is linux and most models with Intel procs should support this server but not sure how to publish a plugin.
#notanissue

RaymondHimle, 2020-02-02:
I can't seem to find any documentation on having a plugin published. You could just create a VM on a QNAP and setup the containers on that.

simonduz, 2020-02-02:
> I can't seem to find any documentation on having a plugin published. You could just create a VM on a QNAP and setup the containers on that.

There is a user supported app store for Qnap separate from the "partner/brand" store.  There is a link on this site for QPKG devs.
[https://www.qnapclub.eu/en](url)

leonardlaw, 2020-05-01:
I built a TSD container myself on a QNAP NAS.  IIRC, I had to pull a package from the qnapclub repo to get git, otherwise, I followed the HOWTO to download and build the container.

simonduz, 2020-05-05:
> I built a TSD container myself on a QNAP NAS. IIRC, I had to pull a package from the qnapclub repo to get git, otherwise, I followed the HOWTO to download and build the container.
This is good news.  I hope to see this come to fruition as an actual Qnap solution.   I'm no developer, I wish help, but I could test if it does become available.

closed 2022-05-14 by kennethjiang