TheSpaghettiDetective/obico-server #166: Email Testing Feature for Stand Alone Server

jreed1701, 2020-02-04:
Issue:
In the context of the stand alone server, there is no method available via command or user interface to validate that the email settings in the [docker-compose.yml](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docker-compose.yml) file are working.  One must initiate a print, simulate a spaghettification, and hope that an email is sent. 

Add a feature to allow a user of the stand alone server to test email configuration. The system shall send a test feature upon activation/initiation using the EMAIL-* under environment in the reference yaml file. 

Reference:
 [docker-compose.yml](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docker-compose.yml) 

Relates to #129 a need for configuration testing.

kennethjiang, 2020-02-08:
Before this issue is solved, a work around is

- Make sure `Notify me when print job is canceled` is checked in `Preferences` page.
- Start a test print job and cancel it.
- Wait for the email to arrive (or not)

jreed1701, 2020-04-26:
@kennethjiang Hi. Sorry to take so long to try this. I did and it did not work. I've got my setting in correctly for sure.  I followed the procedure you suggested but no email.

Furthermore, since this issue card was created the user preferences page has a Verify Email setting. When I click "Re-send Verification" I get a 500 server error. This corroborates the the system is unable to send the email.  

Exact Route; http://IP_ADDR:3334/accounts/email/

I will say, I am using a gmail account with 2FA enabled. Would that cause any issues?

Screen shot for good measure:

[image]

kennethjiang, 2020-04-27:
Yeah SMTP server is a thorny thing. We are paying for AWS SES to get email delivered which is quite reliable.

Please join our private server discord channel at: https://discord.gg/UFH3JC. Hopefully you can find help there

handsome1987, 2020-05-10:
I have the same problem als below.
Does anyone how to fix this?

> @kennethjiang Hi. Sorry to take so long to try this. I did and it did not work. I've got my setting in correctly for sure. I followed the procedure you suggested but no email.
>
> Furthermore, since [...]

> [image]

jagr200, 2020-07-26:
> I have the same problem als below.
> Does anyone how to fix this?

Yes. Change the settings in the file and then rebuild the container using run "sudo docker-compose up --build -d"

goeland86, 2020-08-13:
I managed to get the email verification going. With 2FA on Gmail you need to create an application password that you feed to docker-compose.

What I'm having an issue with is that I got the verification email, but I did not receive any of the job completed or job failed emails I was expecting to receive...

RaymondHimle, 2020-08-13:
This would be a separate issue. However I have noticed this as well and it seems to be a bug. What platform are you running the TSD server on?

goeland86, 2020-08-13:
Ubuntu 20.04 with latest docker-ce.

goeland86, 2020-08-13:
I'm happy to open a separate issue if that helps.

RaymondHimle, 2020-08-13:
X86 or arm?

goeland86, 2020-08-13:
x86 - it's my workshop Intel NUC. Not super powered, but I do have 16GB of ram and 4 logical cores (2 with HT)

closed 2020-08-16 by kennethjiang

kennethjiang, 2020-08-17:
See https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docs/email_guide.md

commit f9606f4d (2020-08-16): close #166