TheSpaghettiDetective/obico-server #167: Include hardware recommendations in docs?

k4kfh, 2020-02-05:
As a new user, I think it would be super helpful to have a rough estimate of how much compute resources we should budget. Is it primarily a memory hog or a CPU hog? Or both? Would the most straightforward way to look at this be "gigabytes of RAM per concurrently-used printer"?

RaymondHimle, 2020-02-05:
No extensive testing has been done on this due to the wide range of CPUs available. It is a CPU workload mainly.

k4kfh, 2020-02-06:
Is there any way I could get a rough idea? I'll have somewhere between 5 and 10 printers I want to spaghetti-detect in the end (although probably not all running concurrently) and I've got some spare PCs at my disposal, but I'm trying to figure out a rough idea of what kind of resources TSD needs.

RaymondHimle, 2020-02-06:
Mid range 3rd gen i5 may be enough. Again we have done no testing on capacity and resources per printer required.

kennethjiang, 2020-02-12:
@k4kfh sorry somehow I missed this issue. @RaymondHimle thank you for chiming in.

TSD is more CPU-bound than IO-bound or memory-bound - to a point. When you have enough load that requires spinning up more gunicorn workers, you will essentially need double among of memory for both `web` and `ml-api`, and maybe `tasks`.

bbrendon, 2020-04-10:
I was running a TSD (single printer) on a system with 4gb RAM that had some other services (relatively light) on it. It OOMed on the first night. So maybe I would guess about 2-3 GB of memory for TSD? Just another datapoint.

kennethjiang, 2020-04-10:
@bbrendon I'm a bit surprised it OOMed on 4GB for only 1 printer. Which process/container OOMed?

On my desktop I set RAM limit to 4GB for docker and it doesn't show any sign of memory shortage.

Check to make sure you have swap set up.

kennethjiang, 2022-05-14:
Done.

closed 2022-05-14 by kennethjiang