TheSpaghettiDetective/obico-server #170: TSD thinks my enclosure is spaghetti

LP0101, 2020-02-10:
I have my printer inside an enclosure with metallic walls. During the early parts of my print, it thinks the walls are spaghetti, and I had to lower the sensitivity to prevent it from alerting me during successful prints.

Here's an example.
[image]

kennethjiang, 2020-02-11:
We are constantly improving the accuracy of our AI model but I have to admit false positives are still more often than I hoped. There are a few common patterns that tend to trick AI into thinking it's a positive. Your picture is one of them.

Hopefully this kind of positives will decrease as we improve the model. There isn't really "a fix" that can magically get rid of it. So I'm closing this bug for now. Let me know if you think otherwise.

closed 2020-02-11 by kennethjiang

LP0101, 2020-02-26:
Just an update here, increasing the lighting in my enclosure solved the issue, no more false positives.

kennethjiang, 2020-02-26:
Thank you for the update. We know we still need to improve the robustness of our AI model. Currently it's quite sensitive to changes of lighting, camera angle, etc.