TheSpaghettiDetective/obico-server #172: Inverse X/Y/Z controls

kennethjiang, 2020-02-16:
DlorangerToday at 10:29 AM
the thing I would like to have is a z invert option.  my anet actually raises the zantry so z up makes sense, but my xy drops the bed so the logic is inverted.

rahools, 2022-08-31:
Hi! I would like to work on the issue. is it up for grabs? Additionally, I am new to open source contributions, thus don't know anything about open source contributions workflows. Nonetheless, I am willing to put in the effort.

kennethjiang, 2022-09-02:
@rahools Thank you for thinking about contributing. Feel free to take a shot. We don't have complicated workflow except [the standard, minimum PR process](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/about-pull-requests). Feel free to submit a PR. I'll let you know if I need you to change anything.

rahools, 2022-09-03:
@kennethjiang submitted a PR #694