TheSpaghettiDetective/obico-server #177: User Preferences leads to Server Error (500) on Stand Alone Server

zcubed11, 2020-03-06:
Hi, I have been trying to get email notifications setup on my stand along server but I can not seem to get emails sent. Going to User Preferences on my server returns a 'Server Error (500)' page. I am not doing anything fancy, just using a normal gmail setup.

This is the output of  docker-compose logs --tail=500 web:

> web_1     | django.request ERROR    Internal Server Error: /user_preferences/
> web_1     | Traceback (most recent call last):
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django_settings_export.py", line 51, in __getitem__
> web_1     |     return super(ExportedSettings, self).__getitem__(item)
> web_1     | KeyError: 'SLACK_CLIENT_ID'
> web_1     |
> web_1     | During handling of the above exception, another exception occurred:
> web_1     |
> web_1     | Traceback (most recent call last):
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/core/handlers/exception.py", line 34, in inner
> web_1     |     response = get_response(request)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/core/handlers/base.py", line 126, in _get_response
> web_1     |     response = self.process_exception_by_middleware(e, request)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/core/handlers/base.py", line 124, in _get_response
> web_1     |     response = wrapped_callback(request, *callback_args, **callback_kwargs)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/contrib/auth/decorators.py", line 21, in _wrapped_view
> web_1     |     return view_func(request, *args, **kwargs)
> web_1     |   File "/app/app/views.py", line 129, in user_preferences
> web_1     |     return render(request, 'user_preferences.html', dict(form=form, bot_name=bot_name))
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/shortcuts.py", line 36, in render
> web_1     |     content = loader.render_to_string(template_name, context, request, using=using)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/loader.py", line 62, in render_to_string
> web_1     |     return template.render(context, request)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/backends/django.py", line 61, in render
> web_1     |     return self.template.render(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 171, in render
> web_1     |     return self._render(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 163, in _render
> web_1     |     return self.nodelist.render(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 937, in render
> web_1     |     bit = node.render_annotated(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 904, in render_annotated
> web_1     |     return self.render(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/loader_tags.py", line 150, in render
> web_1     |     return compiled_parent._render(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 163, in _render
> web_1     |     return self.nodelist.render(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 937, in render
> web_1     |     bit = node.render_annotated(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 904, in render_annotated
> web_1     |     return self.render(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/loader_tags.py", line 62, in render
> web_1     |     result = block.nodelist.render(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 937, in render
> web_1     |     bit = node.render_annotated(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 904, in render_annotated
> web_1     |     return self.render(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/defaulttags.py", line 512, in render
> web_1     |     return self.nodelist.render(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 937, in render
> web_1     |     bit = node.render_annotated(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 904, in render_annotated
> web_1     |     return self.render(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/defaulttags.py", line 302, in render
> web_1     |     match = condition.eval(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/defaulttags.py", line 875, in eval
> web_1     |     return self.value.resolve(context, ignore_failures=True)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 671, in resolve
> web_1     |     obj = self.var.resolve(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 796, in resolve
> web_1     |     value = self._resolve_lookup(context)
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 829, in _resolve_lookup
> web_1     |     current = current[bit]
> web_1     |   File "/usr/local/lib/python3.6/site-packages/django_settings_export.py", line 61, in __getitem__
> web_1     |     .format(key=item)
> web_1     | django_settings_export.UnexportedSettingError: The `SLACK_CLIENT_ID` setting key is not accessible from templates: add "SLACK_CLIENT_ID" to `settings.SETTINGS_EXPORT` to change that.
> web_1     | django.channels.server ERROR    HTTP GET /user_preferences/ 500 [0.37, 172.18.0.1:53692]
> web_1     | django.channels.server INFO     HTTP POST /api/v1/octo/pic/ 200 [13.12, 172.18.0.1:53678]
> web_1     | django.channels.server INFO     HTTP GET /media/tsd-pics/raw/2/5/1583511811.257673.jpg 200 [0.22, 172.18.0.2:47572]
> web_1     | django.channels.server INFO     HTTP POST /api/v1/octo/pic/ 200 [15.16, 172.18.0.1:53696]
>

zcubed11, 2020-03-06:
Is there a format that must be followed in the EMAIL_HOST configs? Like the information must be in quotes or single quotes?

kennethjiang, 2020-03-07:
This looks like a bug we introduced recently. Will fix it soon.

kennethjiang, 2020-03-08:
Just fixed it. Please pull and try it again @zcubed11 . Thank you for reporting this bug.

commit a32720bf (2020-03-08): close #177
closed 2020-03-08 by kennethjiang