TheSpaghettiDetective/obico-server #178: Error 500 on email verification

flyswatta, 2020-03-08:
I don't receive any emails and get error 500 on email verification. Set up using Gmail, using my regular gmail account and password, and recently changed to using app specific password.  Still no change. 
Here's my email config (names/passwords changed to protect the innocent :)
 environment:
    EMAIL_HOST: smtp.gmail.com     # -> such as smtp.gmail.com
    EMAIL_HOST_USER: gobbledegook@gmail.com   # -> such as your email address for a Gmail account
    EMAIL_HOST_PASSWORD: password   # -> your email account password
    EMAIL_PORT: 587
    EMAIL_USE_TLS: 'True'
    DEFAULT_FROM_EMAIL: 'gobbledegook@gmail.com'
    DEBUG: 'False'    # Don't set DEBUG to True, otherwise the static files will be cached in browser until hard-refresh
    SITE_USES_HTTPS: 'False' # set it to 'True' if https is set up
    SOCIAL_LOGIN: 'False'
    REDIS_URL: 'redis://redis:6379'
    DATABASE_URL: 'sqlite:////app/db.sqlite3'
    INTERNAL_MEDIA_HOST: 'http://web:3334'
    ML_API_HOST: 'http://ml_api:3333'
    ACCOUNT_ALLOW_SIGN_UP: 'False'  # -> set to 'True' if you wan

[TSP-Error500.txt](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/files/4303891/TSP-Error500.txt)

kennethjiang, 2020-03-09:
Other users have reported similar issues before. I think it's because Gmail sometimes would reject SMTP for some reasons who don't understand (it has a lot of black magics behind the scene). So Gmail works for some users but not for others.

If you can get a real SMTP server, you should try that to see if it works. If it works then it's just Gmail. Please post your result here.

HeyRobb, 2020-03-13:
Same issue here... Other (less) secure apps have no issues sending with identical settings (587/TLS) from this account.

kennethjiang, 2020-03-14:
If you can, please debug it to see if this is a code error. I can't reproduce it on my side unfortuntately.

kennethjiang, 2020-04-18:
Closing due to inactivity.

closed 2020-04-18 by kennethjiang

erik434, 2020-04-18:
Thanks Kenneth for the lead! Just commenting here for anyone else who hits this problem and comes here from google. I was getting a 500 error when trying to verify my email address on a self-hosted instance as well. Turns out I needed to enable less-secure access on my google account first, and then it was able to send the verification email no problem. I also needed to change "https" to "http" in the email verification link, otherwise it would time out because I don't have https set up for my instance. Hope this helps someone in the future :)

kennethjiang, 2020-04-21:
Thanks @erik434  for the tip!

leonardlaw, 2020-04-21:
One additional suggestion to add to @erik434 is to use an App Password in your Google Account instead of making it "less secure". These are at:

https://myaccount.google.com/apppasswords

Tim-Dixon, 2020-07-12:
@leonardlaw 

I checked out the link you posted and I was informed that "The setting you are looking for is not available for your account.". Is this something that's only available for G-Suite?

erik434, 2020-07-12:
@Tim-Dixon You might only have that option if you've enabled 2-Step Verification on your Google account. Doesn't require G-Suite; I just have a regular free Google account and with 2SV enabled I'm able to create app-specific passwords.

MaxBec, 2020-08-24:
Hey guys,

what do i need to do after changing the .yml settings for my mail account?

TheOddPerson, 2020-12-31:
Error 500 on sending email verification will happen if the email preferences are not entered correctly in the .yml file before starting the server. I'm very new to docker and not certain when the .yml file is checked. What I did was; stopped the WEB container, changed the email settings accordingly as docker environment variables and restarted.

timur-tabi, 2022-05-15:
I'm seeing this problem as well, on two different configurations (one a Jetson Nano, and the other Oracle Cloud).