TheSpaghettiDetective/obico-server #179: Black screen from MotionEye camera mjpeg output

NotExpectedYet, 2020-03-09:
Hello! Just been playing with your self hosted version and I've hit a bit of a road block. 

My streams come from a MotionEye installation on another box due to the camera's been RTSP streams originally. Motioneye converts those to mjpeg and Octoprint picks them up fine from their URL. Motion Eye offers a few different stream url's but the main ones with /frame/1/picture don't show up in octoprint. The normal ones so "http://192.168.1.5:5004" work fine in Octoprint but I don't see an output in Spaghetti detective. 

[image]

I'm sure I'm been an idiot somewhere, but I've double checked the logs and there is no issue with spaghetti detective contacting octoprint that I can see and i've made sure the plugin is pointing at my spaghetti detective local instance. 

Oh! I've also tried with premium streaming turned on and off which result in the same. 

The only thing I haven't done is sign up for an account, or any of that as I presumed been self hosted it wouldn't need it. Let me know if that isn't that case. 

Thanks.

kennethjiang, 2020-03-09:
Have you gone through the steps in https://www.thespaghettidetective.com/docs/webcam-feed-is-not-showing/ and https://www.thespaghettidetective.com/docs/octoprint-is-offline/ ?

NotExpectedYet, 2020-03-10:
Yes, I'm at the last step asking for help. If you look at my picture you'll see they're connected, detecting temps and printing. Both cameras work fine in OctoPrint currently. 

I checked the logs initially as I develop with octoprint so I have debugging on. I don't have time at the moment to get the logs but I will if it helps. There we're no errors though.

kennethjiang, 2020-03-11:
Sorry for the delay. Did you check the self-diagnostic report in OctoPrint TSD settings? https://www.thespaghettidetective.com/docs/octoprint-is-offline/#step-2-check-the-plugins-connection-to-the-spaghetti-detective-server

NotExpectedYet, 2020-03-11:
I'll double check for you, I presumed wrongly that was just for testing connection between the server and octoprint plugin. I'll get back to you, no rush either i'm quite busy at the moment aha.

RaymondHimle, 2020-07-22:
This should be fixed in the latest versions with compatibility mode

closed 2020-07-22 by RaymondHimle