TheSpaghettiDetective/obico-server #180: default octoprint view: thespaghettidetective is loading youtube (and subsequently doubleclick)

Mirarkitty, 2020-03-10:
Tracing a spurious 

_Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://googleads.g.doubleclick.net/pagead/id. (Reason: CORS request did not succeed)._

I find that this URL is loaded by the default Octoprint interface (early, too):
https://www.youtube.com/embed/ZpDw9mNBngU

Would it be possible to only access third party sites on request?

RaymondHimle, 2020-03-10:
See https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/issues/51#issuecomment-589490327

kennethjiang, 2020-03-11:
Thank you @RaymondHimle for chiming in.

@Mirarkitty I'm not attached to YouTube. Just wanted to put a video somewhere visible to help users understand what TSD is. Will be happy to merge your PR if you can figure out another way to do it.

Mirarkitty, 2020-03-21:
Maybe a video link would be enough instead of an embedded video? Not sure why it loads in the default interface, before everything else, because it certainly doesn't display there.

Mirarkitty, 2020-03-21:
[image]
[image]

kennethjiang, 2020-03-21:
This is how OctoPrint works - it loads every links from itself all at once, even if some pages/tabs are not being displayed. For instance, you can see that it's loading "slicing", even if you are not using slicer on it at all.

justinmbrock, 2020-04-04:
Just installed this. And now I'm uninstalling it. Thanks anyway, but I don't need iframes loading your trackers in the background.

Mirarkitty, 2020-04-17:
I'm also pretty sure blaming Octoprint for loading trackers is not a way forward.

kennethjiang, 2020-04-18:
Closing since this is more a philosophical debate than a bug.

closed 2020-04-18 by kennethjiang

Mirarkitty, 2020-04-26:
Wait, forcing a load of a tracker is a philosophical debate? Good to know where you stand.