TheSpaghettiDetective/obico-server #184: High CPU usage by Django, fix found

Kllrv, 2020-03-25:
I discovered Django using unreasonable amounts of CPU, in the range of 25-35% while no print was active. This topic was helpful for me: https://superuser.com/questions/867067/djangos-dev-server-causes-permanent-high-cpu-load-with-python-3

in the docker-compose.yml, I changed the Django launch line:

`command: sh -c "python manage.py collectstatic --noinput && python manage.py migrate && python manage.py runserver --noreload 0.0.0.0:3334"`

the server works as normal, but idle cpu use by TSD processes dropped to negligible amounts.

kennethjiang, 2020-03-25:
Thank you @Kllrv

closed 2020-03-25 by kennethjiang

timothyjhanson, 2022-11-08:
What worked for me was to pip install watchdog and then to use 
`
watchmedo auto-restart -d . -R -p '*.py' -- python manage.py runserver --noreload 0.0.0.0:8000`

commit 79f91fda (2020-03-25): Closes #184