TheSpaghettiDetective/obico-server #185: Trouble connecting TSD-Plugin to Server

leonardlaw, 2020-03-29:
Hi.

Running TSD in a containerized deployment on a home server.  Tried to connect to it for an Octoprint plugin, but I get constant failed to connect to server errors.

Octoprint.log suggests a websocket error:
2020-03-28 17:06:14,041 - octoprint.plugins.thespaghettidetective - DEBUG - Establishing WS connection...
2020-03-28 17:06:16,048 - octoprint.plugins.thespaghettidetective - ERROR - Backing off 130.011645 seconds: Failed to connect to websocket server
2020-03-28 17:06:18,971 - octoprint.plugins.thespaghettidetective - ERROR - Server websocket is closing

TSD-Tasks logs:
django.channels.server INFO     WebSocket HANDSHAKING /ws/dev/ [192.168.86.56:35646]
django.channels.server INFO     WebSocket DISCONNECT /ws/dev/ [192.168.86.56:35646]
daphne.server ERROR    Exception inside application: [Errno -3] Try again
  File "/usr/local/lib/python3.6/site-packages/channels/sessions.py", line 179, in __call__
    return await self.inner(receive, self.send)
  File "/usr/local/lib/python3.6/site-packages/channels/middleware.py", line 41, in coroutine_call
    await inner_instance(receive, send)
  File "/usr/local/lib/python3.6/site-packages/channels/consumer.py", line 59, in __call__
    [receive, self.channel_receive], self.dispatch
  File "/usr/local/lib/python3.6/site-packages/channels/utils.py", line 59, in await_many_dispatch
    await task
  File "/usr/local/lib/python3.6/site-packages/channels_redis/core.py", line 429, in receive
    real_channel
  File "/usr/local/lib/python3.6/site-packages/channels_redis/core.py", line 484, in receive_single
    index, channel_key, timeout=self.brpop_timeout
  File "/usr/local/lib/python3.6/site-packages/channels_redis/core.py", line 324, in _brpop_with_clean
    async with self.connection(index) as connection:
  File "/usr/local/lib/python3.6/site-packages/channels_redis/core.py", line 820, in __aenter__
    self.conn = await self.pool.pop()
  File "/usr/local/lib/python3.6/site-packages/channels_redis/core.py", line 70, in pop
    conns.append(await aioredis.create_redis(**self.host, loop=loop))
  File "/usr/local/lib/python3.6/site-packages/aioredis/commands/__init__.py", line 175, in create_redis
    loop=loop)
  File "/usr/local/lib/python3.6/site-packages/aioredis/connection.py", line 113, in create_connection
    timeout)
  File "/usr/local/lib/python3.6/asyncio/tasks.py", line 339, in wait_for
    return (yield from fut)
  File "/usr/local/lib/python3.6/site-packages/aioredis/stream.py", line 24, in open_connection
    lambda: protocol, host, port, **kwds)
  File "/usr/local/lib/python3.6/asyncio/base_events.py", line 750, in create_connection
    infos = f1.result()
  File "/usr/local/lib/python3.6/concurrent/futures/thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.6/socket.py", line 745, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
  [Errno -3] Try again

Plugin claims token is good, but when setting up printer in TSD, there is an infinite wait for the plugin to connect.

Any ideas?

RaymondHimle, 2020-03-29:
We would need the full debug log. Also did you set the Django site up correctly?

leonardlaw, 2020-03-29:
Here's the octoprint.log (attached).

Django site should be setup correct, to my knowledge.  Just setup the site
and a user, per the screenshot attached.

leonardlaw, 2020-03-29:
[image]
[octoprint (1).log](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/files/4399438/octoprint.1.log)

RaymondHimle, 2020-03-30:
We need the debug logging turned on then reboot the device please.

leonardlaw, 2020-03-30:
[octoprint (4).log](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/files/4400461/octoprint.4.log)

kennethjiang, 2020-03-30:
@leonardlaw  if you run `docker-compose log web`, do you see any errors?

leonardlaw, 2020-03-30:
Ah yes. Lots of errors.  Looks like a Docker networking thing?  It's trying
to get to port 35942, but only 3334 is open in the docker-compose.yml

web_1     | django.channels.server INFO     WebSocket HANDSHAKING /ws/dev/ [
192.168.86.56:35942]

web_1     | django.channels.server INFO     WebSocket DISCONNECT /ws/dev/ [
192.168.86.56:35942]

web_1     | daphne.server ERROR    Exception inside application: [Errno -3]
Try again

web_1     |   File
"/usr/local/lib/python3.6/site-packages/channels/sessions.py", line 179, in
__call__

web_1     |     return await self.inner(receive, self.send)

web_1     |   File
"/usr/local/lib/python3.6/site-packages/channels/middleware.py", line 41,
in coroutine_call

web_1     |     await inner_instance(receive, send)

web_1     |   File
"/usr/local/lib/python3.6/site-packages/channels/consumer.py", line 59, in
__call__

web_1     |     [receive, self.channel_receive], self.dispatch

web_1     |   File
"/usr/local/lib/python3.6/site-packages/channels/utils.py", line 59, in
await_many_dispatch

web_1     |     await task

web_1     |   File
"/usr/local/lib/python3.6/site-packages/channels_redis/core.py", line 429,
in receive

web_1     |     real_channel

web_1     |   File
"/usr/local/lib/python3.6/site-packages/channels_redis/core.py", line 484,
in receive_single

web_1     |     index, channel_key, timeout=self.brpop_timeout

web_1     |   File
"/usr/local/lib/python3.6/site-packages/channels_redis/core.py", line 324,
in _brpop_with_clean

web_1     |     async with self.connection(index) as connection:

web_1     |   File
"/usr/local/lib/python3.6/site-packages/channels_redis/core.py", line 820,
in __aenter__

web_1     |     self.conn = await self.pool.pop()

web_1     |   File
"/usr/local/lib/python3.6/site-packages/channels_redis/core.py", line 70,
in pop

web_1     |     conns.append(await aioredis.create_redis(**self.host,
loop=loop))

web_1     |   File
"/usr/local/lib/python3.6/site-packages/aioredis/commands/__init__.py",
line 175, in create_redis
[... 222 lines cut ...]
web_1     |     return (yield from fut)

web_1     |   File
"/usr/local/lib/python3.6/site-packages/aioredis/stream.py", line 24, in
open_connection

web_1     |     lambda: protocol, host, port, **kwds)

web_1     |   File "/usr/local/lib/python3.6/asyncio/base_events.py", line
750, in create_connection

web_1     |     infos = f1.result()

web_1     |   File "/usr/local/lib/python3.6/concurrent/futures/thread.py",
line 56, in run

web_1     |     result = self.fn(*self.args, **self.kwargs)

web_1     |   File "/usr/local/lib/python3.6/socket.py", line 745, in
getaddrinfo

web_1     |     for res in _socket.getaddrinfo(host, port, family, type,
proto, flags):

web_1     |   [Errno -3] Try again

kennethjiang, 2020-03-30:
any idea where `192.168.86.56:35946` could come from?

leonardlaw, 2020-03-30:
I am pretty sure that's the Octoprint server.

kennethjiang, 2020-03-31:
No TSD plugin doesn't need to connect to OctoPrint server.

Did you follow this guide to set up TSD plugin: https://www.thespaghettidetective.com/docs/octoprint-plugin-setup/ ?

leonardlaw, 2020-03-31:
Yes, I followed the guide.  The process hangs when in the Add Printer flow
in TSD, it is basically waiting for connection "forever".  This is
corroborated by a number of connection errors in the TSD-plugin
troubleshooting flow.

kennethjiang, 2020-03-31:
ok. where did you enter octoprint ip/port in? TSD plugin settings?

leonardlaw, 2020-03-31:
Correct, TSD plugin settings - I have an override set to http://192.168.86.55:3334/ in the advanced settings tab.  Verifies that "token is good", but from the logs, seems like the WebSockets are connecting over different ports.

kennethjiang, 2020-03-31:
Sorry I'm quite confused. What I asked was `192.168.86.56:35946`, not `192.168.86.55:3334`.

At this point this feels like a configuration issue, not a bug. And it's not very efficient to debug the configuration issue via github.

If you have some server debugging skills, please do it yourself or get other people to help you with it. And come back to us when/if you confirm this is a bug.

We do offer consultation for private TSD server if you need us to be more involved in helping you figure this out. However we will need to charge for the service, since we are a small team and we don't have enough resource to offer consultation to all free users.

leonardlaw, 2020-03-31:
To be clear:

192.168.86.55 is the server running TSD.  It is receiving connections on 3334 as per the ports opened in the docker-compose.yml.

192.168.86.56 is the raspberry pi running Octoprint and the TSD plugin.  It appears to be trying to access the TSD server via websockets and getting blocked.

What I'm suggesting is that the docker networking setup is missing something that is causing this connectivity lapse.

leonardlaw, 2020-03-31:
And.. it's solved.  Needed to add the default network to the container networking section for the web container.

closed 2020-03-31 by leonardlaw

FrozenAtlas, 2020-04-26:
Hey, @leonardlaw I'm having the same issues with the WebSockets connecting all over the place. I see you managed to fix it but how exactly did you fix it?

leonardlaw, 2020-05-01:
So, for me, when I originally built my container, the network was set to "traefik_proxy" which is kind of like a VLAN that proxies everything through traefik.  I had to add the "default" network to my docker-compose.yaml in order to get it talk to the plugin, which is naturally running on a rpi.

nikolak, 2020-08-30:
I also have this same problem, and can't quite figure it out.

I've only entered the email relevant env in the docker compose, the connection from octoprint works.

This is what I see in the web container logs

```
api.consumers WARNING  WebConsumer: Closed websocket with code: None
django.channels.server INFO     WebSocket HANDSHAKING /ws/web/2/ [10.0.1.1:58259]
django.channels.server INFO     WebSocket CONNECT /ws/web/2/ [10.0.1.1:58259]
django.channels.server INFO     WebSocket DISCONNECT /ws/web/2/ [10.0.1.1:58259]
api.consumers WARNING  WebConsumer: Closed websocket with code: None
```

The port keeps changing, and also I'm seeing it try to connect to docker internal IPs:

```
api.consumers WARNING  WebConsumer: Closed websocket with code: None
django.channels.server INFO     WebSocket HANDSHAKING /ws/web/2/ [172.23.0.1:57200]
django.channels.server INFO     WebSocket CONNECT /ws/web/2/ [172.23.0.1:57200]
django.channels.server INFO     WebSocket DISCONNECT /ws/web/2/ [172.23.0.1:57200]
api.consumers WARNING  WebConsumer: Closed websocket with code: None
```

`10.0.1.1` is my laptop, and `10.0.10.1` is where the TSD is running.

Sometimes it works for a bit, but then just stops after octoprint restart, or just by itself.