TheSpaghettiDetective/obico-server #186: ModuleNotFoundError: No module named 'config'

thorgod, 2020-03-31:
Web & Tasks keeps restarting running docker locally. 

[image]

[image]

thorgod, 2020-03-31:
Here is the web. 
docker-compose logs -f web

[image]

kennethjiang, 2020-03-31:
Please run `docker-compose exec web sh`, and you will be dropped into a shell. In that shell, please run the follow commands and post the output here (please use copy/past, not screenshots):

`pwd`
`pip list`

thorgod, 2020-03-31:
I cant even get into shell before it restarts. Maybe its because I am running a windows 7, looking at the video it sounds like you need windows 10?

[image]

RaymondHimle, 2020-03-31:
Sounds like the app resources are not in the container. Can you rebuild the container with ```docker-compose --build -d```

thorgod, 2020-03-31:
@RaymondHimle 
I assume you mean  `docker-compose up --build -d` because  `docker-compose --build -d` gives me the help options

(Text https://pastebin.com/kZF3GLDb)
[image]

thorgod, 2020-04-01:
Its ok I just created a docker image on AWS Cloud and it there no problems. I was just hoping to run it locally first.

closed 2020-04-01 by thorgod