TheSpaghettiDetective/obico-server #187: Exec format error

Firestorm7893, 2020-04-05:
I was trying to install thespaghettidevice on an nvidia jetson nano, since in theory it is made to run neural networks, But when i try to use docker i get an "exec format error". Is it because the jetson is an arm device?

This is the output i get (sorry for the bad formatting but i'm still learning how github works):

`sudo docker-compose up -d

Creating network "thespaghettidetective_default" with the default driver

Pulling redis (redis:5.0-alpine)...

5.0-alpine: Pulling from library/redis

8a0637ca1ac9: Pull complete

97ca613f2fd6: Pull complete

80d356bfcf65: Pull complete

7524782d4455: Pull complete

3505be5efd10: Pull complete

fe6596326ebd: Pull complete

Digest: sha256:511275ddf4c4582e296f6fca803df853a8ab2b733a704c44c498f0a6a90e5ba0

Status: Downloaded newer image for redis:5.0-alpine

Building tasks

Step 1/7 : FROM thespaghettidetective/web:base-1.1

base-1.1: Pulling from thespaghettidetective/web

bdf0201b3a05: Pull complete

59c926705abf: Pull complete

dd4853571cc7: Pull complete

3a45eb710779: Pull complete

ace61d116e35: Pull complete

c587cdbc1ad6: Pull complete

8a8616bf140a: Pull complete

076c9514f459: Pull complete

e4293e5bdf0e: Pull complete

2126aebcaff4: Pull complete

8e3389e1339c: Pull complete

Digest: sha256:49acc31d36caeefa9f34aa1c3c6d54fb9b3fab24cd1c942c8c12470aa8a4ea2e

Status: Downloaded newer image for thespaghettidetective/web:base-1.1

 ---> 71fa3a4936ec

Step 2/7 : WORKDIR /app

 ---> Running in 1023d7e60ede

Removing intermediate container 1023d7e60ede

 ---> c1d7c1fbd41d

Step 3/7 : EXPOSE 3334

 ---> Running in e635db8a405b

Removing intermediate container e635db8a405b

 ---> 61d1c099a126

Step 4/7 : ADD . /app

 ---> 85f6d8619a58

Step 5/7 : RUN pip install -U pip

 ---> Running in 93d948304785

standard_init_linux.go:211: exec user process caused "exec format error"

ERROR: Service 'tasks' failed to build: The command '/bin/sh -c pip install -U pip' returned a non-
zero code: 1`

RaymondHimle, 2020-04-05:
Please follow the advanced doc https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docs/jetson_guide.md

RaymondHimle, 2020-04-06:
@kennethjiang this can be considered resolved

kennethjiang, 2020-04-09:
Thanks @RaymondHimle

closed 2020-04-09 by kennethjiang