TheSpaghettiDetective/obico-server #191: Octoprint addon - Own Server - HTTPS Lockup

MidnightLink, 2020-04-17:
Hosting my own TSD server.

If you have https put in instead of http when adding your own server address and click save it will freeze the TSD plugin/octoprint and the server will need to be rebooted.

kennethjiang, 2020-04-17:
@MidnightLink this is probably caused by some networking setup. TSD production is https so the code obviously works with https.

Please dig a bit deeper and post more details here. Please come to our private server discord channel you need assistance on how to debug this problem: https://discord.gg/Cn9Hde

RaymondHimle, 2020-04-18:
I could not reproduce this issue in my test environment.

RaymondHimle, 2020-05-06:
@kennethjiang This can be closed due to inactivity.

closed 2020-05-06 by kennethjiang