TheSpaghettiDetective/obico-server #192: [Help Requested] Setting up on UnRAID

litecactus, 2020-05-04:
I want to run this on my unraid box with the builtin docker but I don't know how. I've tested it on Windows Docker and it all installs via the instructions just great, but the setup looks complicated for a manual process. Is there a way of running that setup from unraid docker directly?

RaymondHimle, 2020-05-04:
You would be best off asking in the TSD Discord. However I an not sure of anyone running TSD in UnRAID.

postfach, 2020-05-14:
If you haven't got it working yet, I think I've just done it.  SSH to your server using PuTTY, then install docker compose with:

`sudo curl -L "https://github.com/docker/compose/releases/download/1.25.5/docker-compose-$(uname -s)-$(uname -m)" -o /usr/local/bin/docker-compose`

then:

 `sudo chmod +x /usr/local/bin/docker-compose`

Now it's just a case of installing as per the TSD instructions - Before you start, navigate to a share (for example /mnt/user/MyShare) (perhaps use your appdata share) 

and run

`git clone https://github.com/TheSpaghettiDetective/TheSpaghettiDetective.git`

then 

`cd TheSpaghettiDetective && sudo docker-compose up -d`

Updating/editing anything will need to be done manually as there is no template like you get when you install a container from CA but it appears to be working at first glance.

[image]

[image]

kennethjiang, 2020-05-14:
Thanks for the update! And glad to know it worked! Closing this issue now.

closed 2020-05-14 by kennethjiang

famewolf, 2020-06-25:
I guess I'm missing the obvious but how do you create an account on your local server? The only option I see is sign in and the credentials from the actual thespaghettidetective website do not work.

jagr200, 2020-07-26:
Has anyone got email to work with this? I can't get it to work and ive updated the docker-compose file.

famewolf, 2020-07-26:
DId you reboot the docker host and the containers?  It didn't work till after I'd done that after rebuilding.

jagr200, 2020-07-26:
I restarted the 4 containers. How do you rebuild them?

jagr200, 2020-07-26:
> DId you reboot the docker host and the containers? It didn't work till after I'd done that after rebuilding.

I restarted the 4 containers. How do you rebuild them?

famewolf, 2020-07-26:
> > DId you reboot the docker host and the containers? It didn't work till after I'd done that after rebuilding.
> 
> I restarted the 4 containers. How do you rebuild them?

Delete the container and rerun the docker compose.   Follow their instructions to do an "update" and it might work also which involves a git put and then the docker compose line.  It's in the instructions.

jagr200, 2020-07-26:
> Delete the container and rerun the docker compose. Follow their instructions to do an "update" and it might work also which involves a git put and then the docker compose line. It's in the [...]

That did it. Thanks man. I've got a print going so I'm not connected to Octoprint right now to make sure everything is working. But should know soon.

jagr200, 2020-07-26:
> Delete the container and rerun the docker compose. Follow their instructions to do an "update" and it might work also which involves a git put and then the docker compose line. It's in the [...]

Any idea how to combine these into one docker image instead of 4?