TheSpaghettiDetective/obico-server #195: Get current status

maziggy, 2020-05-11:
AFAIK there is no API for TSD, but is there any way to get the current status from remote?

Thanks, Martin

kennethjiang, 2020-05-11:
We are currently collecting requirements to decide if we should provide public APIs, and if so, what APIs.

Can you elaborate your use cases that will require TSD API(s)?

maziggy, 2020-05-11:
I just finished at QT based dashboard application for my printers which I run on a small RPI with touch display. Would be nice to also add the TSD status to the dashboard.

Thanks, Martin

kennethjiang, 2020-05-17:
Sorry for the delay!

I see your use case now. However, I'm not sure how common this use case is. Let's keep this issue open to see if there are more users who want the similar thing so that we can justify spending time to provide the API for it.

cmsimike, 2020-06-02:
I'd like an API if possible. 
My use case is pretty simple: I would like a more direct (and self hosted way) to be notified if a print goes wrong using my home automations.

My home is very automated, from a kodi in every room, to  snips.ai running in rooms (which allows local voice control), more RGB lights than I'd like to publicly admit, as well as a lot of other things all connected and all powered by home assistant.

I'd like home assistant have the knowledge of whether or not the print has failed. Once home assistant knows, I can set up automations such as:
- If the print fails, pause the media playing on any Kodi (under the assumption that if media is playing, I am listening and/or watching) and use text to speech to tell me directly the print failed
- Currently I am pretty terrified to leave a print going overnight while I'm sleeping. If I had an API the automations can detect if I'm in bed (which it already can) and if a print failure happens while I'm asleep I can automatically turn on the bedroom lights and generate a loud noise to wake me up.
- I can ask snips how the print job is doing and it can give me feed back on where the meter (
[image] this thing) is.
- Using the same info as above, I can be alerted at any % not when it hits red. If it gets high enough in the yellow, let's say, I can notify myself to go take a look. Maybe it is something I can try to resolve before it becomes a bigger problem.
- If home assistant detects a failure, home assistant can figure out which room I'm in and announce the print seems to have failed and ask whether or not I want to pause the print. I can figure out ways of getting camera data to me.
- Use iOS notifications (again through home assistant) to be notified on my phone (with a camera screen shot) if the print has failed and use actionable notifications in ios to cancel the job.

There are probably way cooler ways to be notified that I haven't thought of yet.

If an api were available (and better yet if it can be two way communication, but I'd be more than happy if the api only exposed if the print looks like it failed or not) then any other device I have connected to my home automations can act as a beacon to notify me.

There are some ways I can accomplish this now but none of which I am too excited to use.
- I currently use the email notification. That's fine, but requires constant vigilance on my email (which I'll be honest with you, isn't always the case).
- I can setup home assistant with an email address, and have TSD email my automation and my automation can take it from there. While a bit round-about, this might be a path I explore if an API doesn't happen. 
- TWILIO, pushbullet, etc - all external services. I try not to rely on external services as best as I can. My apartment automations, I'd say, are 95% functional if I were to disconnect the internet.

Thanks for your consideration!

t413, 2020-10-27:
I'd like this too!

At least exposing the spaghetti factor _back to the octoprint_ instance so I can read it from plugins, MQTT, and by extension my home automation / MQTT hardware projects.

wernight, 2020-11-03:
I'd also welcome a way (any way) to include the meter in Home Assistant). I like MQTT but pretty much anything would work.

Spirtogram, 2020-11-03:
I also use octoprint and home assistant. all information is collected on one card. this is print data, plugin enclosure control (on-off, light, fan, box heat). I want to add a spaghetti sensor. home assistant can notify about events without third-party applications. api wants to be simple. request, json response. as octoprint did.

kennethjiang, 2022-05-14:
public API is supported now: https://www.obico.io/docs/api/

closed 2022-05-14 by kennethjiang