TheSpaghettiDetective/obico-server #200: Timelapse wont play or Download

secco04, 2020-05-29:
Hi there
Ive never got it working to play my timelapse within TSD.
When i Klick download /play nothing happens 
In the Docker logs i See some 404 errors. 
The mp4 files are working properly when downloading via scp... 

`10]
web_1     | django.channels.server INFO     HTTP GET /static/img/3d_printer.5afb4300522d.png 200 [0.01, 192.168.2.101:47436]
web_1     | django.channels.server INFO     HTTP GET /static/img/detective-hour-inverse.9ea6129acfef.png 200 [0.01, 192.168.2.101:47434]
web_1     | django.channels.server INFO     WebSocket HANDSHAKING /ws/web/2/ [192.168.2.101:47448]
web_1     | django.channels.server INFO     WebSocket CONNECT /ws/web/2/ [192.168.2.101:47448]
web_1     | django.channels.server INFO     WebSocket HANDSHAKING /ws/janus/2/ [192.168.2.101:47452]
web_1     | django.channels.server INFO     WebSocket CONNECT /ws/janus/2/ [192.168.2.101:47452]
web_1     | django.channels.server INFO     WebSocket DISCONNECT /ws/janus/2/ [192.168.2.101:47452]
web_1     | django.channels.server INFO     WebSocket DISCONNECT /ws/web/2/ [192.168.2.101:47448]
web_1     | api.consumers WARNING  WebConsumer: Closed websocket with code: 1001
web_1     | api.consumers WARNING  WebConsumer: Closed websocket with code: 1001
web_1     | django.channels.server INFO     WebSocket HANDSHAKING /ws/web/2/ [192.168.2.101:47526]
web_1     | django.channels.server INFO     WebSocket CONNECT /ws/web/2/ [192.168.2.101:47526]
web_1     | django.channels.server INFO     WebSocket HANDSHAKING /ws/janus/2/ [192.168.2.101:47528]
web_1     | django.channels.server INFO     WebSocket CONNECT /ws/janus/2/ [192.168.2.101:47528]
web_1     | django.channels.server INFO     WebSocket DISCONNECT /ws/janus/2/ [192.168.2.101:47528]
web_1     | django.channels.server INFO     WebSocket DISCONNECT /ws/web/2/ [192.168.2.101:47526]
web_1     | api.consumers WARNING  WebConsumer: Closed websocket with code: 1001
web_1     | api.consumers WARNING  WebConsumer: Closed websocket with code: 1001
web_1     | django.channels.server INFO     HTTP GET /prints/ 200 [0.06, 192.168.2.101:47438]
web_1     | django.request WARNING  Not Found: /prints/None
web_1     | django.channels.server WARNING  HTTP GET /prints/None 404 [0.00, 192.168.2.101:47538]
web_1     | django.request WARNING  Not Found: /prints/None
web_1     | django.channels.server WARNING  HTTP GET /prints/None 404 [0.00, 192.168.2.101:47538]
web_1     | django.request WARNING  Not Found: /prints/None
web_1     | django.channels.server WARNING  HTTP GET /prints/None 404 [0.00, 192.168.2.101:47538]
web_1     | django.channels.server INFO     HTTP GET /prints/ 200 [0.05, 192.168.2.101:48176]
web_1     | django.channels.server INFO     HTTP GET /static/hijack/hijack-styles.a7cafac63295.css 200 [0.00, 192.168.2.101:48176]
web_1     | django.request WARNING  Not Found: /prints/None
web_1     | django.channels.server WARNING  HTTP GET /prints/None 404 [0.00, 192.168.2.101:48198]
web_1     | django.channels.server INFO     HTTP GET /static/img/logo-inverted.d0cd99b8b560.png 200 [0.00, 192.168.2.101:48198]
web_1     | django.request WARNING  Not Found: /prints/None
web_1     | django.channels.server WARNING  HTTP GET /prints/None 404 [0.00, 192.168.2.101:48204]
web_1     | django.request WARNING  Not Found: /prints/None
web_1     | django.channels.server WARNING  HTTP GET /prints/None 404 [0.00, 192.168.2.101:48204]
root@spaghetti-detective:/home/detect`

kennethjiang, 2020-06-01:
Sorry for the delay.

From the log it looks like somehow the link the web page send `/prints/None` in the request. Can you use Javascript console to debug why this is the case?

secco04, 2020-06-04:
Some how its looking for the timelapses at "Example.Com..."
I cant find the misleading configuration file or entry

secco04, 2020-06-05:
my fault - i deleted all timelapses and printed one with the correct site name and the urls were fine

closed 2020-06-05 by secco04

Austin3dPrinting, 2020-06-16:
Hey there, I'm having the same experience as secco04 but Ive checked my site name and it seems correct to me. I'm attaching a screen vid of the settings and time-lapse issue. I can attach the logs also but I'm not sure where to find them.
[TheSpaghettiDetecive Timelapse Issue.mov.zip](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/files/4788202/TheSpaghettiDetecive.Timelapse.Issue.mov.zip)

secco04, 2020-06-16:
Can you check the console output of the Javascript at your Browser if the video URLs are correct

Austin3dPrinting, 2020-06-16:
Weirdly enough I went to Chrome (from Safari) to look at that and it works fine. It must be something with how Safari loads the page 🤷🏼‍♂️

kennethjiang, 2020-06-16:
That is strange. I'm not aware that download doesn't work in Safari...

Thanks @secco04  for helping people around!

Austin3dPrinting, 2020-06-16:
The download actually works just fine in Safari it looks like it is just the time-lapse viewing / loading in the browser. You can see it in the [screen grab attached above](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/files/4788202/TheSpaghettiDetecive.Timelapse.Issue.mov.zip)^

Yes thank you @secco04 for the help!

secco04, 2020-06-17:
@kennethjiang @Austin3dPrinting no problem thats what community is for ;)
But since i dont own a Mac i cant help out more... 
Maybe Safari is blocking the js execution in some way?