TheSpaghettiDetective/obico-server #201: Newest commits won't build (95c9f166b712cbfa2bf1100b878c30ddbd6595bd)

derekslenk, 2020-06-01:
Current master at commit 95c9f166b712cbfa2bf1100b878c30ddbd6595bd won't run. It will build fine but upon load,  only get a 500 server error.

In the logs:
```
django.request ERROR    Internal Server Error: /accounts/login/,
Traceback (most recent call last):,
  File "/usr/local/lib/python3.6/site-packages/django/core/handlers/exception.py", line 34, in inner,
    response = get_response(request),
  File "/usr/local/lib/python3.6/site-packages/django/core/handlers/base.py", line 145, in _get_response,
    response = self.process_exception_by_middleware(e, request),
  File "/usr/local/lib/python3.6/site-packages/django/core/handlers/base.py", line 143, in _get_response,
    response = response.render(),
  File "/usr/local/lib/python3.6/site-packages/django/template/response.py", line 106, in render,
    self.content = self.rendered_content,
  File "/usr/local/lib/python3.6/site-packages/django/template/response.py", line 83, in rendered_content,
    content = template.render(context, self._request),
  File "/usr/local/lib/python3.6/site-packages/django/template/backends/django.py", line 61, in render,
    return self.template.render(context),
  File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 171, in render,
    return self._render(context),
  File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 163, in _render,
    return self.nodelist.render(context),
  File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 937, in render,
    bit = node.render_annotated(context),
  File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 904, in render_annotated,
    return self.render(context),
  File "/usr/local/lib/python3.6/site-packages/django/template/loader_tags.py", line 150, in render,
    return compiled_parent._render(context),
  File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 163, in _render,
    return self.nodelist.render(context),
  File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 937, in render,
    bit = node.render_annotated(context),
  File "/usr/local/lib/python3.6/site-packages/django/template/base.py", line 904, in render_annotated,
    return self.render(context),
  File "/usr/local/lib/python3.6/site-packages/django/template/library.py", line 192, in render,
    output = self.func(*resolved_args, **resolved_kwargs),
  File "/app/app/templatetags/vue.py", line 27, in bundle,
    url = static(f'{STATIC_BUNDLE_PREFIX}/{ext}/{bundle_name}.{ext}'),
  File "/usr/local/lib/python3.6/site-packages/django/templatetags/static.py", line 167, in static,
    return StaticNode.handle_simple(path),
  File "/usr/local/lib/python3.6/site-packages/django/templatetags/static.py", line 118, in handle_simple,
    return staticfiles_storage.url(path),
  File "/usr/local/lib/python3.6/site-packages/django/contrib/staticfiles/storage.py", line 153, in url,
    return self._url(self.stored_name, name, force),
  File "/usr/local/lib/python3.6/site-packages/django/contrib/staticfiles/storage.py", line 132, in _url,
    hashed_name = hashed_name_func(*args),
  File "/usr/local/lib/python3.6/site-packages/django/contrib/staticfiles/storage.py", line 420, in stored_name,
    raise ValueError("Missing staticfiles manifest entry for '%s'" % clean_name),
ValueError: Missing staticfiles manifest entry for 'frontend/js/main.js',
django.channels.server ERROR    HTTP GET /accounts/login/?next=/printers/ 500 [0.21, 192.168.12.100:59572],
django.request WARNING  Not Found: /favicon.ico,
```

Going back to bf3e4e0 and TSD works fine.

I believe the issue is with the static build changes in https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/commit/d91ee90d4a6ec1881235a4544f6df2cba2d91ba8

kennethjiang, 2020-06-01:
Thank you @derekslenk for reporting this bug! I reverted d91ee90 for now. I still know too little about Vue to mess around. ;)

Please pull and let me know if it's fixed now.

derekslenk, 2020-06-01:
> 
> 
> Thank you @derekslenk for reporting this bug! I reverted [d91ee90](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/commit/d91ee90d4a6ec1881235a4544f6df2cba2d91ba8) for now. I still know too little about Vue to mess around. ;)
> 
> Please pull and let me know if it's fixed now.

Working again, thanks for the quick response!

closed 2020-06-01 by derekslenk