TheSpaghettiDetective/obico-server #203: [FEATURE REQUEST] Indicator for Maximum spaghetti detected on a print

serenewaffles, 2020-06-04:
It would be nice if TSD indicated the maximum spaghetti detected on a print using the gauge already implemented. I imagine this happening similar to mechanical max reading gauges.

As can be seen in the below image, the black hand represents the current reading. When it moves higher, it catches the red hand and moves it as well. The red hand is not spring loaded, so it does not return. I imagine having a small triangle on the outer part of the TSD dial to indicate it as well. This could help users determine whether review is necessary for ongoing prints. Please see the second picture for an example.

[image]

[image]

kennethjiang, 2020-06-05:
Really good suggestion.

What will be a good way to help new users understand the small triangle means the max value?

serenewaffles, 2020-07-01:
If we're concerned about how a new user would understand, I recommend either a help text popup (on hover) or gating it behind an option. Another option could be to have a quick popup for new users (or for all users after this is implemented) that has a simple picture and explanation.

kennethjiang, 2020-07-01:
I guess I don't know how to articulate the benefit of this new piece of info to users. In other words, I'm struggling to figure out how we can tell users something like "this 'max detection' means X. You can use it to do Y. Take Z action when this value is too high/low."

serenewaffles, 2020-07-01:
Having played more with Focused Feedback, I'm no longer certain this addition offers any benefit. Focused Feedback is already alerting users to frames that TSD is unsure of.

closed 2020-07-01 by serenewaffles