TheSpaghettiDetective/obico-server #205: getting "Server Error (500)" for any page other than the /admin address after configuration.

lukeglazebrook, 2020-06-06:
Installed TSD fine a few times on Ubuntu blew the container and everything away and started fresh.  After the initial config I am now getting "Server Error (500)" for all sites excluding the /admin page.  I have even gone to the extent of rebuilding with a freh Ubuntu box, reinstalling Docker etc and TSD with the same results.  Has something changed?  Feel free to educate me if I am doing something stupid.

lukeglazebrook, 2020-06-06:
Obviously configured the sites page, any idea where the logs are that I need to examine?

[image]

lukeglazebrook, 2020-06-06:
Sad times ...

[image]

kennethjiang, 2020-06-06:
@lukeglazebrook Can you run `docker-compose logs web` and look for exceptions in the output?

kennethjiang, 2020-06-06:
@lukeglazebrook Just pushed a commit. Can you do `git pull && docker-compose restart` and let me know if you are still getting 500 errors?

matzebecker, 2020-06-06:
I had the same issue like @lukeglazebrook and your last commit solved the problem for me, @kennethjiang! Thanks!

kennethjiang, 2020-06-06:
Awesome! Will close the issue now.

closed 2020-06-06 by kennethjiang

lukeglazebrook, 2020-06-06:
Well done Ken mate you fixed it, didn't even need a restart.  I must say I have never used this Github thingy I think its got some real potential to take off big

war3zlod3r, 2020-06-26:
having this same issue I can only access the pages from localhost, 127.0.0.1 or my local ip won't work and return error 500

cdvankammen, 2025-06-24:
this seems to be an issue I am still getting and have tried everything under the sun with docker web commands. Any thoughts on why this is not accessbile from outside of local host?