TheSpaghettiDetective/obico-server #217: Web_1 keeps restarting

Tumtum2814, 2020-06-11:
Trying to setup the detective on windows 10 pro, went smooth, but web_1 keeps restarting.
Getting this from the logs. 
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"
web_1     | standard_init_linux.go:211: exec user process caused "no such file or directory"

kennethjiang, 2020-06-11:
Can you attach the full log output for `web` container?

Tumtum2814, 2020-06-11:
maybe I'm missing something, But it's literally the only message that web outputs for forever.  Both from the Docker Dashboard and from docker-compose logs -f
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
[... cut ...]
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"
standard_init_linux.go:211: exec user process caused "no such file or directory"

kennethjiang, 2020-06-11:
if you run `docker-compose stop` followed by `docker-compose up`, you should get the full output.

Tumtum2814, 2020-06-11:
[startup log.txt](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/files/4762491/startup.log.txt)
ah gotcha, lots more to chew on lol. Attached.

kennethjiang, 2020-06-11:
https://futurestud.io/tutorials/how-to-fix-exec-user-process-caused-no-such-file-or-directory-in-docker

This may be relevant. Although I know most Windows users are able to run it without any problem.

Tumtum2814, 2020-06-12:
Wow. You are a master. That was the fix. She's up and going now. Thanks so much!

closed 2020-06-12 by Tumtum2814