TheSpaghettiDetective/obico-server #224: Secret token URL

azreal1983, 2020-06-16:
Hello

Process:
On setting up a new server with server address http://<MYIP>:3334/ as instructed with a proceeding / the URL on setup tab:

step 3: Copy the secret token to the box below.
If you have problem locating the secret token, click here:
The here URL hyperlink is: http://<MYIP>:3334//secure_redirect/?target=/printer_auth_token/0/&source=plugin

with 2 // following the 334.

RaymondHimle, 2020-06-16:
Some more info would be helpful. What is the exact problem here? And what does your DJANGO site setup look like?

azreal1983, 2020-06-17:
The exact problem is when you click on the here hyperlink the URL has 2 "/" after the 3334 rather than 1.

RaymondHimle, 2020-06-17:
What does your Django site URL look like https://github.com/TheSpaghettiDetective/TheSpaghettiDetective#configure-django-site

azreal1983, 2020-06-17:
http://<MYIP>:3334/admin/

RaymondHimle, 2020-06-17:
Do not enter the trailing slash in the Server Address field in the plugin. This will be updated in the docs

RaymondHimle, 2020-06-17:
@kennethjiang This looks to be related to the new feature you are working on

RaymondHimle, 2020-06-17:
Closing in favor of https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/issues/65

closed 2020-06-17 by RaymondHimle

kennethjiang, 2020-07-17:
Sorry for taking so long to get to this issue. This is more of an issue with outdated documentation, when we changed the code so that it doesn't require trailing /. Fixed in TheSpaghettiDetective/TheSpaghettiDetective@5798674