TheSpaghettiDetective/obico-server #229: No time-lapse

gknops, 2020-06-22:
I installed TSD using docker on macOS. It is working, I can see the stream, it sends emails. But it does not appear to record timelapses. How can I debug this?

Thanks for TSD!

kennethjiang, 2020-06-22:
Can you run `docker-compose logs tasks` and post the output here?

gknops, 2020-06-23:
The logs looked fine, that made me try a different browser. On Chrome the time lapses show, on Safari they just show an ever-spinning animation. Also the download-button doesn't render:

[image]

kennethjiang, 2020-06-23:
Hmmm this looks like a javascript/css compatibility problem in different browser.

We are rewriting the time-lapse page. Please wait for a few more days as I don't want to debug a page that will be retired soon.

gknops, 2020-06-23:
Reasonable. Thank you!

RaymondHimle, 2020-07-22:
Closing due to inactivity

closed 2020-07-22 by RaymondHimle

MathijsG, 2023-12-13:
I have the same issue. When I do `docker-compose logs tasks` I see:

tasks_1   | [2023-12-13 14:02:58,142: ERROR/ForkPoolWorker-2] Task app.tasks.compile_timelapse[06f500b1-f931-44a7-90d8-50ccb75e24d9] raised unexpected: FileNotFoundError(2, "No such file or directory: 'ffmpeg'")

Does ffmpeg needs to be installed on the host device itself?

tam-thread, 2024-02-17:
I have an identical issue as above - log is the same except date and time - I running on Ubuntu and when my print finishes I'll install ffmpeg if it isn't there already

MathijsG, 2024-02-18:
You need to have ffmpeg within the container, not in the host per se.

With the help from the support from Discord they pointed me to do the following command:

`` docker run --rm -it obico-server_web which ffmpeg```

What does it report?

If you only get an empty line, instead of a path to ffmpeg, then I'd suggest you try to rebuild the image with:

``docker-compose up --build``

dralkaar, 2025-01-19:
I have the same error: "The media could not be loaded, either because the server or network failed or because the format is not supported."
From what i could find out using the mobile app the problem is that it redirects wrong. 

It uses: https://localhost:3334/linktovideo instead of https://hostname/linktovideo

MathijsG, 2025-01-19:
> I have the same error: "The media could not be loaded, either because the server or network failed or because the format is not supported." From what i could find out using the mobile app the [...]

How is your domain setup? https://www.obico.io/docs/server-guides/configure/

3334 is the default port, if you change that you also need to change that in Django.

dralkaar, 2025-01-20:
Thanks for the quick reply. 
I managed to make it work properly.
The domain setup was correct (but i did remove the localhost:3334 since i can always add it via command line), the only thing that i was missing was the proper .env file.  For future reference (this is something that is not clear in the documentation) when the obico is using a dns (accessible via the internet) the .env file needs to be created and must have the proper values in it:

DJANGO_SECRET_KEY=
SITE_IS_PUBLIC=True
CSRF_TRUSTED_ORIGINS=["https://hostname.domain", "https://*.hostname"]
SOCIAL_LOGIN= 
ACCOUNT_ALLOW_SIGN_UP=
ADMIN_IP_WHITELIST=["ipthatcanlogintoadminsite"]

These are the ones that i've configured.
SOCIAL_LOGIN and ACCOUNT_ALLOW_SIGN_UP are not mandatory but i use them since i want to restrict the access and registration.

kennethjiang, 2026-01-31:
> > Better yet, if you’re open to it, I’d be happy to merge a PR if you want to update the docs directly! Since I wrote the app, I sometimes miss the "first-time user" perspective, and your input would be invaluable.
> 
> Sure, I can do a documentation PR with what I think makes sense. Although I would probably question here if it isn't more of a bug that the domain for timelapse storage is seemingly hardcoded into the individual timelapses without being able to refresh them?
> 
> That would essentially mean that if I move the server from my current domain to a new domain, then all my past timelapses would not be available on the webpage.
> 
> As right now I experience the following when trying to view/download timelapses. This is all while I am connected to the website at `https://mydomain.dev`
> 
> **Timelapse made before I set the site to public in `.env`**
> 
> <img alt="Image" width="2000" height="1012" src="https://private-user-images.githubusercontent.com/7889925/542966083-c27b0a11-1d16-49a4-a3ea-44fea41e7e82.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Njk4MjA0NDcsIm5iZiI6MTc2OTgyMDE0NywicGF0aCI6Ii83ODg5OTI1LzU0Mjk2NjA4My1jMjdiMGExMS0xZDE2LTQ5YTQtYTNlYS00NGZlYTQxZTdlODIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI2MDEzMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNjAxMzFUMDA0MjI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTRjM2Y5OTJhMDRhMDNhZmZhOTgxNDIxM2I5NDI0NWEwNzVkZWMzOTY4YjUxZGI3ZTczYWRhOWI2MmYyZWQzOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.aRLi9MT0UxvqthN9tvG3AzdbFucDAnvYYpkgrdeIDIw">
> **Timelapse made after I set the site to public in `.env`**
> 
> <img alt="Image" width="2000" height="1077" src="https://private-user-images.githubusercontent.com/7889925/542966960-f042745e-1442-42d8-8ccc-c813e34efad1.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Njk4MjA0NDcsIm5iZiI6MTc2OTgyMDE0NywicGF0aCI6Ii83ODg5OTI1LzU0Mjk2Njk2MC1mMDQyNzQ1ZS0xNDQyLTQyZDgtOGNjYy1jODEzZTM0ZWZhZDEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI2MDEzMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNjAxMzFUMDA0MjI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTFkZDYyZThkZjg2YWUwNGVlYzhhMjM2MTdmMTQ3YTM0ZGI0NDBjZWJhODMzYjllZmM0ZWJkMDk3YjI3NGYxYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.oxGVHuY0v-unb8KE_nBgh8f2a1rrcVM--4jHmQMC1wY">
> My expectation was that either I would be able to config what domain to try and use to pull video storage if its on a different server, or it would use the URL the website is currently being visited by, so `https://mydomain.dev` as the network request.
> 
> I might be missing something or not knowing about a usecase here, just seems a bit odd that it does not use the url the website is visited/hosted on. :)
> 
> Also, thanks for having this open-sourced and available to selfhost! Saved me a few failed prints already!! 👍

Another developer has tried to solve this problem, but the issue is that in the cloud, we don't host the media files on the same servers as the web server. So we cannot assume that they have the same domain name and hence only stored the relative URL.   for Obico cloud, they were not even stored on a single server. It was stored in the cloud storage.  and we need to have a solution that can support this use case too. But I will be happy to accept your PR if you can come up with an elegant solution.

Crowdedlight, 2026-01-31:
> Another developer has tried to solve this problem, but the issue is that in the cloud, we don't host the media files on the same servers as the web server. So we cannot assume that they have the [...]

So you could have different server/domain names per timelapse as it gets split between servers? 
Yeah, I see the challenge then. An `.env` config where you set the address of the server storing the timelapses would not help either then, as it could change per timelapse I assume? 

I have not checked your codebase but I guess one possibility for us that use a single server, would be a tool that could be run like the `manage.py` to replace the domain of all past timelapses. So we could edit to get past timelapses working if moving domain. 
But might be out of scope for you to officially include/support and be something we can hack on our own. ;-) 

For now I will just do a PR on the documentation as you suggested to include a warning/info that when changing domain/site the effect only works for new timelapses.

Crowdedlight, 2026-01-31:
Made a PR and added the python script I made to replace all my old timelapse paths with my new domain. Verified to work on my server and can now watch the earlier timelapses from my new domain. 

#1129