TheSpaghettiDetective/obico-server #242: [QUESTION] Self Hosted Servers and TSD Feedback

serenewaffles, 2020-07-01:
I'm not sure if this is the right place to ask, but I couldn't find any information elsewhere.

I am currently hosting my own instance of TSD locally. I use it to monitor my prints and I try to look at and evaluate all timelapses.

When I provide feedback, am I actually helping the greater TSD community?

If not, is there a way I can use the timelapses my hosted server generates to help the greater TSD community?

If it turns out I have to upload those timelapses to the TSD website, is there a way I could donate my DH to e.g. someone using it to make masks?

kennethjiang, 2020-07-01:
Thank you for opening this issue. I just added some docs to answer some of your questions: https://www.thespaghettidetective.com/docs/help-the-detective-improve/

Detective Hours are not transferable. So you can't donate your DHs to some other users. But if you write to us `support at thespaghettidetective dot com`, we can do it for you on a one-off basis. :)

serenewaffles, 2020-07-01:
This is a very good start, thank you! Could you please clarify whether going through the feedback process on my server uploads the videos (e.g. when I do focused feedback and give TSD permission to look at a video, does that video then get uploaded?).

kennethjiang, 2020-07-01:
Focused Feedback is only for videos that are already in TSD cloud. Basically the videos need to be uploaded to TSD before they are eligible for Focused Feedback (still being implemented).

serenewaffles, 2020-07-01:
I checked the updated page and it covers all my questions, thanks!

closed 2020-07-01 by serenewaffles

marcelrv, 2021-12-22:
> Focused Feedback is only for videos that are already in TSD cloud. Basically the videos need to be uploaded to TSD before they are eligible for Focused Feedback (still being implemented).

Sorry to revive this old discussion with a clarification question/confirmation.

From the https://www.thespaghettidetective.com/docs/help-the-detective-improve/ page and my general understanding wrt running a private TSD server:
> There is no purpose of clicking the `Did she get it right?` this information is not shared with TSD, and my local model does not learn for it. Right?
> There is no purpose of using the  `Focused Feedback` button as again  this information is not shared with TSD.

To make the Focussed Feedback usable, I would need to manually upload the videofile to the TSD hosted version, or the feature would need to be improved with some sort of consent button to upload the file automaticcaly to the TSD cloud along with the feedback.

Absent of such a upload feature, does it make sense to suppress these buttons if it is a private TSD server, as they don't serve a purpose?

kennethjiang, 2021-12-23:
> Absent of such a upload feature, does it make sense to suppress these buttons if it is a private TSD server, as they don't serve a purpose?

This is a good point. The only argument against that is there are private server owners who want to have a way to collect the data from their own server and do their own training.

An option I can think of is to create an installation script that asks "do you want to enable focused feedback" when the server is installed for the first time and store this option somewhere in the config. But it obviously adds a lot of complexity to the project that may not be justified by the benefits.