TheSpaghettiDetective/obico-server #244: Unable to configure Twilio SMS

RobeeeJay, 2020-07-02:
I hope I'm doing something silly, I can't get Twilio SMS to be an option. When I try to configure it I just get this above the checkbox:

`Please configure TWILIO_* items in settings to enable phone alert`

This is my `docker-compose.override.yml` file:

```
version: '2.4'

x-web-defaults: &web-defaults
  build:
    dockerfile: Dockerfile.base
  environment:
    TWILIO_ACCOUNT_SID: 'ACxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
    TWILIO_AUTH_TOKEN: '78xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx'
    TWILIO_FROM_NUMBER: '+441234123456'

services:
  ml_api:
    build:
      context: ml_api
      dockerfile: Dockerfile.aarch64
    environment:
        HAS_GPU: 'True'
    runtime: nvidia
```

I then run `docker-compose up --build -d` and it is updating the containers. I've also tried restarting the containers manually, deleting them and rebuilding.

Does anyone know what I might be doing wrong?

kennethjiang, 2020-07-02:
What do you mean by "I can't get Twilio SMS to be an option."? There is no extra option for Twilio beside the lines you have changed.

Do you mean you still can't receive SMS after the lines are configured correctly?

RobeeeJay, 2020-07-03:
Thanks for the response! When I go to configure my phone number so it knows where to send them, this is what I see:

[image]

Which implies I've not set the env vars properly, or I'm doing something else wrong. :(

kennethjiang, 2020-07-04:
From your screenshot it does look like at least one fo the `TWILIO_*` settings are missing.

Can you run `docker-compose exec web bash` to shell in the container, and then run `env | grep TWILIO_` to check if they are set up correctly?

RobeeeJay, 2020-07-04:
Hmm no they aren't, I don't know how they aren't being picked up by the docker compose override file, I will try and find out...

kennethjiang, 2020-07-04:
Not sure if `docker-compose.override.yml` is somehow not honored. Maybe try to put these env vars directly in `docker-compose.yml`and try it again?

RobeeeJay, 2020-07-04:
I tried them in `docker-compose.yml` and they worked, which is cool. I'm not sure why they are ignored in the override. Googling for an answer as to why isn't helping either. :(

It looks like an `.env` file might work but I'll need to try it. Although it takes so long to build a new container on the Jetson Nano this might take me some time to give it a go. :)

At least I can enter my mobile number now, thank you!

kennethjiang, 2020-07-04:
It sounds like the docker-compose on Jetson behaves differently when handling `docker-compose.override.yml` file. Good to know `docker-compose.yml` still works. Closing it now.

closed 2020-07-04 by kennethjiang

crkochan, 2021-02-11:
I found that I had to create blocks specifically for the `web` and `tasks` services in the override file when trying to specify email service settings. Trying to add an `x-web-defaults` block to file did not seem cause docker-compose to apply it to the containers.

##### docker-compose.override.yml
```yaml
version: '2.4'

services:
  web:
    environment:
      EMAIL_HOST: 'smtp.gmail.com'
      EMAIL_HOST_USER: 'foo@bar.xyz'
      EMAIL_HOST_PASSWORD: 'hunter2'
      DEFAULT_FROM_EMAIL: 'tsd@bar.xyz'

  tasks:
    environment:
      EMAIL_HOST: 'smtp.gmail.com'
      EMAIL_HOST_USER: 'foo@bar.xyz'
      EMAIL_HOST_PASSWORD: 'hunter2'
      DEFAULT_FROM_EMAIL: 'tsd@bar.xyz'
```

kennethjiang, 2021-02-11:
it looks like you already used a stub username/password. but I still redacted the passwords just to be sure.

crkochan, 2021-02-12:
Yes, those were [bogus entries](http://bash.org/?244321).