TheSpaghettiDetective/obico-server #245: Add Dockerfile to Unraid Community Applications

Archmonger, 2020-07-05:
Community Applications is like a "docker app store" for Unraid users. I'm sure the entire Unraid community would be thrilled if you added your dockerfile to this.

[Community Applications Link](https://forums.unraid.net/topic/38582-plug-in-community-applications/)

kennethjiang, 2020-07-06:
Thank you for submitting this issue. Sounds like a great idea!

Unfortunately none of the TSD core team members has the skills or time at this moment to work on this. But if anyone can submit a PR, we will be happy to merge it.

Archmonger, 2020-11-21:
Seems like a docker exists. However, it's missing any information on what environment variables are needed, and what ports are used by default.

[https://hub.docker.com/r/thespaghettidetective/web](https://hub.docker.com/r/thespaghettidetective/web)

kennethjiang, 2020-11-26:
Sorry for the delay. We have been very swamped by the mobile app launch.

What you found is the base image for the containers. It is not supposed to have env vars defined. The env vars are defined in https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docker-compose.yml.

Archmonger, 2020-12-25:
I found an individual on youtube who created a tutorial on setting it up on Unraid. Unfortunately he uses some unconventional methods to get things running that most users wouldn't want.

[https://youtube.com/watch?v=B2gjxL0MgEo](https://youtube.com/watch?v=B2gjxL0MgEo)

SNR-Tech-Bytes, 2021-01-20:
> I found an individual on youtube who created a tutorial on setting it up on Unraid. Unfortunately he uses some unconventional methods to get things running that most users wouldn't want.
> 
> https://youtube.com/watch?v=B2gjxL0MgEo

Hey that's me! I've been trying to figure out how to get it on the app store, but unfortunately it seems that the deployment of docker compose seems to make it incompatible with the standard app deployments. I believe it would take a rewrite in order to get it to work with the regular apps on unraid. Take this statement with a grain of salt as I'm much more of a hardware engineer than software engineer so there is probably something I'm missing but that was my conclusion while making that video

Archmonger, 2021-01-20:
Hey SNR,

I would assume it's possible to rewrite it to use standard docker deployment methods (rather than docker-compose). However, this would most likely require creating a new docker image. I'm not an expert on Docker but I do know of one who might be able to give some advice @Roxedus

Roxedus, 2021-01-20:
The multi-container nature makes it unsuited for CA

Archmonger, 2021-01-20:
Technically speaking, would there be limitations on moving everything needed into one container?

kennethjiang, 2021-01-20:
The best practice for container is one container for each process. Since TSD server needs multiple processes (web server, db, redis, batch job, etc), we went with multiple containers.

I'm not sure I 100% understand why it's desirable put everything in 1 container. But an option is to use VM, not container to achieve that. Vagrant is a bit old technology but may be something that's worth trying.

kennethjiang, 2021-01-20:
> Hey that's me! I've been trying to figure out how to get it on the app store, but unfortunately it seems that the deployment of docker compose seems to make it incompatible with the standard app [...]

Thank you so much for the awesome video!

Roxedus, 2021-01-20:
> The best practice for container is one container for each process. Since TSD server needs multiple processes (web server, db, redis, batch job, etc), we went with multiple containers.
> 
> 
> 
> I'm not sure if I 100% understand why it's desirable put everything in 1 container. But an option is to use VM, not container to achieve that. Vagrant is a bit old technology but may be something that's worth trying.

It's more about how unraid does things, the biggest caveat is the networking(not being able to use container names for resolving, or internal routing).

kennethjiang, 2021-01-20:
> It's more about how unraid does things, the biggest caveat is the networking(not being able to use container names for resolving, or internal routing).

It does sound we need a unraid-specific `Dockerfile`, and/or `docker-compose.yml`.

Archmonger, 2021-01-20:
Unraid does support VMs but they are hugely inefficient and far less convenient. So most users prefer to use dockers when possible.

scyto, 2021-09-23:
> Gotcha. It does sound we need a unraid-specific `Dockerfile`, and/or `docker-compose.yml`.

Not really, the issue is the docker-compose file being used here is very creative but not (IMHO) the 'best' way for docker container pipelines to work when used for community.  I am having issues figuring out how to use portainer for the same reasons.  I suspect this is partly due to how long ago this was created. (yes best is subjective :-).  Here are some suggestions:

1. move build process for TSD image to a github buildx action - this can be used to build multi architecture builds that can then be pushed to docker hub (advantage, precompiled, deterministic, easily tested, doesn't take any build time for a new user deploying,  doesn't require a new user to understand git and the whole code tree.

2. create new compose and or stack file that uses image(s) above and can easily be reused for unraid, synology, portainer, whatever 'simplified' docker system folks are looking for (pulling the whole git down for portainer is a PITA and hard to manage in a swarm)

Is this of interest?  If so i am happy to take a stab it (i have done it before) in a fork and then propose changes?

scyto, 2021-09-23:
@kennethjiang the issue is less one of multiple containers (compose quite happily creates services even on unraid); the issue i believe is more of creating everything at first runtime.  for example i have the same issue with wanting to put this in a docker swarm / use portainer to manage it.

Ideally the build of each image should be done offline (say using github actions - which has advantage of also using buildx for multi-arch builds). and then using the compose file purely to pull the images and run them as docker services (or via a docker stack).  

This would enable easy usage across a number of different approaches to docker / docker platforms.

kennethjiang, 2021-09-25:
@scyto Good point about using buildx to prebuild multi-arch images. Feel free to send a PR for that!

scyto, 2021-09-25:
Lol, will do, may take me a few weeks to look at this, most of my effort is wrangling support folks of who makes my 3d printer.
I may need help working through the logic of base images for each service, not sure if github discussions is enabled on this repo?

kennethjiang, 2021-09-26:
I'll respond a lot more quickly if you ping me on our Discord server: https://discord.gg/hsMwGpD. I'm the "Kenneth" there - shouldn't be hard to find.