TheSpaghettiDetective/obico-server #247: Docker build fails

BSass, 2020-07-13:
When I try to build the web container, it fails in step 9/20:

> Step 9/20 : RUN if [[ "$uid" != 0 ]]; then     (addgroup --gid $gid $group || echo "group with gid $gid exists") &&     adduser --disabled-password --uid $uid --ingroup $(cat /etc/group|grep ":$gid:" |cut -d: -f1) $user &&     mkdir -p /home/$user /home/$user/.cache /home/$user/.ipython /home/$user/.local &&     mkdir -p /app/ /app/frontend /app/static_build &&     chown -R $uid:$gid /app /home/$user     ; fi
>  ---> Running in 702ca5db1869
> addgroup: option requires an argument: gid
> group with gid  exists
> BusyBox v1.29.3 (2019-01-24 07:45:07 UTC) multi-call binary.
> 
> Usage: addgroup [-g GID] [-S] [USER] GROUP
> 
> Add a group or add a user to a group
> 
>         -g GID  Group id
>         -S      Create a system group
> BusyBox v1.29.3 (2019-01-24 07:45:07 UTC) multi-call binary.
> 
> Usage: adduser [OPTIONS] USER [GROUP]
> 
> Create new user, or add USER to GROUP
> 
>         -h DIR          Home directory
>         -g GECOS        GECOS field
>         -s SHELL        Login shell
>         -G GRP          Group
>         -S              Create a system user
>         -D              Don't assign a password
>         -H              Don't create home directory
>         -u UID          User id
>         -k SKEL         Skeleton directory (/etc/skel)
> ERROR: Service 'web' failed to build: The command '/bin/sh -c if [[ "$uid" != 0 ]]; then     (addgroup --gid $gid $group || echo "group with gid $gid exists") &&     adduser --disabled-password --uid $uid --ingroup $(cat /etc/group|grep ":$gid:" |cut -d: -f1) $user &&     mkdir -p /home/$user /home/$user/.cache /home/$user/.ipython /home/$user/.local &&     mkdir -p /app/ /app/frontend /app/static_build &&     chown -R $uid:$gid /app /home/$user     ; fi' returned a non-zero code: 1
> 

I tried to pull it again to a new folder and then ran 
`docker system prune -f `
before a new try, but the error comes again.

Thanks

encetamasb, 2020-07-14:
Hi! 

Do you get this output when building with

    docker-compose build

command? Or do you set any build args?

Please try

    docker-compose build --build-arg uid=0 --build-arg gid=0

Does this solve the issue?

BSass, 2020-07-14:
Thanks for your reply. I used 

`docker-compose up --build -d`

like the manual suggests. Using 

`docker-compose build --build-arg uid=0 --build-arg gid=0`

raises the same error and shows this extra warning:

`[Warning] One or more build-args [uid gid] were not consumed`

encetamasb, 2020-07-14:
I did touch last time the dockerfile
and would like to help. Your
error is rather strange for me.

Could you tell me your OS/docker/docker-compose versions? Do you have any active override (existing docker-compose.override.yml)? What’s the hash of the commit you are trying to build?

It’s enough to test building web image/container only.

BSass, 2020-07-14:
I run docker on a Synology NAS (x86).

`docker-compose version 1.26.2, build eefe0d31
Docker version 18.09.8, build bfed4f5
`

I used the latest version from git: a69d078f23ce3facd1b247cf4b37aab3f2c02a03
I do not use a docker-compose.override.yml.

encetamasb, 2020-07-14:
Could you use the following Dockerfile for an other test? Changed arg names a bit and it prints what uid/gid is set for the build.

https://gist.github.com/encetamasb/1aadb5df633f83f751e0de06aac30558

You need to put that into web/

Or you can find this in a branch of mine.

https://github.com/encetamasb/TheSpaghettiDetective/tree/encetamasb/docker-uid-gid-issue

encetamasb, 2020-07-14:
This is an older version of the file what must work:

https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/a4cd41fe71c2faa241b0c7eb61a13edd6849960c/web/Dockerfile

.. if you would like to get it work asap.  I would be happy if you could report back whether the arg changes above fixes the issue or not.

BSass, 2020-07-14:
Thank you for your effort. I cloned your branch tried to built it without success:

`Building web
Step 1/20 : FROM thespaghettidetective/web:base-1.1
 ---> 60302abcde4f
Step 2/20 : ARG uid=0
 ---> Using cache
 ---> e47b3ced346a
Step 3/20 : ARG gid=0
 ---> Using cache
 ---> 994398bccc26
Step 4/20 : ARG user=root
 ---> Using cache
 ---> 35679a30de93
Step 5/20 : ARG group=root
 ---> Using cache
 ---> 86d6307ae948
Step 6/20 : ARG with_node=0
 ---> Using cache
 ---> e3ff92a9588b
Step 7/20 : ARG with_sqlite=0
 ---> Using cache
 ---> 4f64de981ba0
Step 8/20 : RUN if [[ "$with_node" == 1 ]]; then     apk add --no-cache --update inotify-tools nodejs npm && npm install -g yarn     ; fi &&     if [[ "$with_sqlite" == 1 ]]; then     apk add --no-cache sqlite     ; fi
 ---> Using cache
 ---> 885f9f4bc163
Step 9/20 : RUN if [[ "$uid" != 0 ]]; then     (addgroup --gid $gid $group || echo "group with gid $gid exists") &&     adduser --disabled-password --uid $uid --ingroup $(cat /etc/group|grep ":$gid:" |cut -d: -f1) $user &&     mkdir -p /home/$user /home/$user/.cache /home/$user/.ipython /home/$user/.local &&     mkdir -p /app/ /app/frontend /app/static_build &&     chown -R $uid:$gid /app /home/$user     ; fi
 ---> Running in 89d48dee2740
addgroup: option requires an argument: gid
BusyBox v1.29.3 (2019-01-24 07:45:07 UTC) multi-call binary.

Usage: addgroup [-g GID] [-S] [USER] GROUP

Add a group or add a user to a group

        -g GID  Group id
        -S      Create a system group
group with gid  exists
BusyBox v1.29.3 (2019-01-24 07:45:07 UTC) multi-call binary.

Usage: adduser [OPTIONS] USER [GROUP]

Create new user, or add USER to GROUP

        -h DIR          Home directory
        -g GECOS        GECOS field
        -s SHELL        Login shell
        -G GRP          Group
        -S              Create a system user
        -D              Don't assign a password
        -H              Don't create home directory
        -u UID          User id
        -k SKEL         Skeleton directory (/etc/skel)
ERROR: Service 'web' failed to build: The command '/bin/sh -c if [[ "$uid" != 0 ]]; then     (addgroup --gid $gid $group || echo "group with gid $gid exists") &&     adduser --disabled-password --uid $uid --ingroup $(cat /etc/group|grep ":$gid:" |cut -d: -f1) $user &&     mkdir -p /home/$user /home/$user/.cache /home/$user/.ipython /home/$user/.local &&     mkdir -p /app/ /app/frontend /app/static_build &&     chown -R $uid:$gid /app /home/$user     ; fi' returned a non-zero code: 1
`

When using the old Dockerfile you provided, it works well. Nice to have it running again :-)

encetamasb, 2020-07-14:
:+1: good to hear the latter!

I see some hope as according your build output you have not used my branch :)

encetamasb, 2020-07-14:
Ok, this is an issue in synology's docker.

    I faced to a similar problem discovered when I tried to do a docker build with ARG declarations.

    ARG values are always empty, whether with a default value in the Dockerfile or 
    a value passed with --build-arg parameter.
    The only workaround is tu use ENV variables to maintain ARG values. 
    It is working but implies to modify any Dockerfile before beeing usable.

https://community.synology.com/enu/forum/1/post/128146?page=4&sort=oldest

I'll pr a fix later this week. @kennethjiang

BSass, 2020-07-14:
Oh sorry, I dont use git very often. I have now manually pasted your Dockerfile with the following result:

`Building web
Step 1/20 : FROM thespaghettidetective/web:base-1.1
 ---> 60302abcde4f
Step 2/20 : ARG TSD_UID=0
 ---> Running in d2a7fea45b9b
Removing intermediate container d2a7fea45b9b
 ---> dfd705839569
Step 3/20 : ARG TSD_GID=0
 ---> Running in 92f6f956d8e8
Removing intermediate container 92f6f956d8e8
 ---> 66d0e54b60e9
Step 4/20 : ARG TSD_USER=root
 ---> Running in 0c66443b7126
Removing intermediate container 0c66443b7126
 ---> ca1808b6d3be
Step 5/20 : ARG TSD_GROUP=root
 ---> Running in a9eae77cfe65
Removing intermediate container a9eae77cfe65
 ---> 34bb140b0775
Step 6/20 : ARG WITH_NODE=0
 ---> Running in 4c86065106ec
Removing intermediate container 4c86065106ec
 ---> 716a27e901e5
Step 7/20 : ARG WITH_SQLITE=0
 ---> Running in ee51383f243c
Removing intermediate container ee51383f243c
 ---> 3f4315a1543a
Step 8/20 : RUN if [[ "${WITH_NODE}" == 1 ]]; then     apk add --no-cache --update inotify-tools nodejs npm && npm install -g yarn     ; fi &&     if [[ "${WITH_SQLITE}" == 1 ]]; then     apk add --no-cache sqlite     ; fi
 ---> Running in 38c103ac88f9
Removing intermediate container 38c103ac88f9
 ---> 3b8f6ddf5bf2
Step 9/20 : RUN echo "User/group configuration: UID: ${TSD_UID} | GID: ${TSD_GID} | USER: ${TSD_USER} | GROUP: ${TSD_GROUP}" &&     if [[ "${TSD_UID}" != 0 ]]; then     (addgroup --gid ${TSD_GID} ${TSD_GROUP} || echo "group with gid ${TSD_GID} exists") &&     adduser --disabled-password --uid ${TSD_UID} --ingroup $(cat /etc/group|grep ":${TSD_GID}:" |cut -d: -f1) ${TSD_USER} &&     mkdir -p /home/${TSD_USER} /home/${TSD_USER}/.cache /home/${TSD_USER}/.ipython /home/${TSD_USER}/.local &&     mkdir -p /app/ /app/frontend /app/static_build &&     chown -R ${TSD_UID}:${TSD_GID} /app /home/${TSD_USER}     ; fi
 ---> Running in 3056d0a35213
User/group configuration: UID:  | GID:  | USER:  | GROUP:
addgroup: option requires an argument: gid
BusyBox v1.29.3 (2019-01-24 07:45:07 UTC) multi-call binary.

Usage: addgroup [-g GID] [-S] [USER] GROUP

Add a group or add a user to a group

        -g GID  Group id
        -S      Create a system group
group with gid  exists
BusyBox v1.29.3 (2019-01-24 07:45:07 UTC) multi-call binary.

Usage: adduser [OPTIONS] USER [GROUP]

Create new user, or add USER to GROUP

        -h DIR          Home directory
        -g GECOS        GECOS field
        -s SHELL        Login shell
        -G GRP          Group
        -S              Create a system user
        -D              Don't assign a password
        -H              Don't create home directory
        -u UID          User id
        -k SKEL         Skeleton directory (/etc/skel)
ERROR: Service 'web' failed to build: The command '/bin/sh -c echo "User/group configuration: UID: ${TSD_UID} | GID: ${TSD_GID} | USER: ${TSD_USER} | GROUP: ${TSD_GROUP}" &&     if [[ "${TSD_UID}" != 0 ]]; then     (addgroup --gid ${TSD_GID} ${TSD_GROUP} || echo "group with gid ${TSD_GID} exists") &&     adduser --disabled-password --uid ${TSD_UID} --ingroup $(cat /etc/group|grep ":${TSD_GID}:" |cut -d: -f1) ${TSD_USER} &&     mkdir -p /home/${TSD_USER} /home/${TSD_USER}/.cache /home/${TSD_USER}/.ipython /home/${TSD_USER}/.local &&     mkdir -p /app/ /app/frontend /app/static_build &&     chown -R ${TSD_UID}:${TSD_GID} /app /home/${TSD_USER}     ; fi' returned a non-zero code: 1
`

BSass, 2020-07-14:
Thank you for your investigation :-)

MaxBec, 2020-08-24:
hey guys,

i have exactly the same problem. is there a fix already?

encetamasb, 2020-08-24:
@MaxBec Hi, hopefully merged soon. You can use the (very same) file linked in the comment if you are in hurry.

> This is an older version of the file what must work:
>
> https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/a4cd41fe71c2faa241b0c7eb61a13edd6849960c/web/Dockerfile
>
> .. if you would [...]

kennethjiang, 2020-08-24:
@BSass @MaxBec just merged the PR. can you test it again?

closed 2020-08-24 by kennethjiang

MaxBec, 2020-08-24:
Yes! The old Dockerfile works! Thank you!

BSass, 2020-08-24:
Works perfectly! Thanks!

PR #274 (2020-08-24): Separated dev Dockerfile to fix compability issues
Should fix #247 

Current docker version on Synology NAS systems is buggy and cannot properly handle build ARG-s.