TheSpaghettiDetective/obico-server #249: docker-compose is not able to create the /web directory: read-only file system

thegillimonster, 2020-07-18:
I'm getting this output:

alex@spaghet:/TheSpaghettiDetective$ sudo docker-compose up -d
thespaghettidetective_redis_1 is up-to-date
thespaghettidetective_ml_api_1 is up-to-date
Starting thespaghettidetective_tasks_1 ... error
Starting thespaghettidetective_web_1   ... 

Starting thespaghettidetective_web_1   ... error

ERROR: for thespaghettidetective_web_1  Cannot start service web: error while creating mount source path '/TheSpaghettiDetective/web': mkdir /TheSpaghettiDetective: read-only file system

ERROR: for tasks  Cannot start service tasks: error while creating mount source path '/TheSpaghettiDetective/web': mkdir /TheSpaghettiDetective: read-only file system

ERROR: for web  Cannot start service web: error while creating mount source path '/TheSpaghettiDetective/web': mkdir /TheSpaghettiDetective: read-only file system
ERROR: Encountered errors while bringing up the project.

I have tried "chmod -R 775 /TheSpaghettiDetective" and that did not help

thegillimonster, 2020-07-18:
I did a lot of googling and figured it out. When you install docker as part of a fresh Ubuntu install by letting the Ubuntu installer do it for you, it installs using snap. Apparently this really mucks up the permissions. I followed this stackoverflow answer and it seems to be working.

https://stackoverflow.com/questions/45764477/docker-compose-error-while-creating-mount-source-path

closed 2020-07-18 by thegillimonster

kennethjiang, 2020-07-18:
Wow this seems like a really tricky problem. Kudos to @thegillimonster for figuring this out yourself!

thegillimonster, 2020-07-18:
> 
> 
> Wow this seems like a really tricky problem. Kudos to @thegillimonster for figuring this out yourself!

Thanks Kenneth :) I'm sure I'll have new problems soon! Way she goes.