TheSpaghettiDetective/obico-server #251: http://169.44.201.108:7002/jetpacks/4.2.1/libopencv_3.3.1-2-g31ccdfe11_arm64.deb connection refused

timur-tabi, 2020-07-19:
I'm following the instructions on installing on a Jetson Nano, and it dies at step 8/13:

Step 8/13 : RUN wget -O /libopencv_3.3.1-2-g31ccdfe11_arm64.deb http://169.44.201.108:7002/jetpacks/4.2.1/libopencv_3.3.1-2-g31ccdfe11_arm64.deb &&         dpkg --install /libopencv_3.3.1-2-g31ccdfe11_arm64.deb
 ---> Running in b160aa7bab75
--2020-07-19 04:42:46--  http://169.44.201.108:7002/jetpacks/4.2.1/libopencv_3.3.1-2-g31ccdfe11_arm64.deb
Connecting to 169.44.201.108:7002... failed: Connection refused.
ERROR: Service 'ml_api' failed to build: The command '/bin/sh -c wget -O /libopencv_3.3.1-2-g31ccdfe11_arm64.deb http://169.44.201.108:7002/jetpacks/4.2.1/libopencv_3.3.1-2-g31ccdfe11_arm64.deb &&   dpkg --install /libopencv_3.3.1-2-g31ccdfe11_arm64.deb' returned a non-zero code: 4

I tested the the URL:

$ wget http://169.44.201.108:7002/jetpacks/4.2.1/libopencv_3.3.1-2-g31ccdfe11_arm64.deb
--2020-07-18 23:51:32--  http://169.44.201.108:7002/jetpacks/4.2.1/libopencv_3.3.1-2-g31ccdfe11_arm64.deb
Connecting to 169.44.201.108:7002... failed: Connection refused.

kennethjiang, 2020-07-19:
It looks like their server has an outage. Hopefully it's temporary. You can try it after a few hours/days.

timur-tabi, 2020-07-19:
I don't know anything about docker.  Would it be possible to run a test on all the servers before starting the installation?

kennethjiang, 2020-07-19:
You mean starting all services without docker and test them? It is possible but it'll take a lot of work to make sure the services can talk to each other the way they are expected to. Since the error you ran into is not specific to Docker, it won't help to try to start the server without docker.

timur-tabi, 2020-07-19:
Ok, well, the site is still down.  What site is is supposed to be, anyway?

$ host 169.44.201.108
108.201.44.169.in-addr.arpa domain name pointer 6c.c9.2ca9.ip4.static.sl-reverse.com.

On a side note, I see a few other errors that don't seem to stop the installation but still look bad:

    Get:64 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main arm64 vim arm64 2:8.0.1453-1ubuntu1.3 [981 kB]
    debconf: delaying package configuration, since apt-utils is not installed
    Fetched 36.9 MB in 12s (3108 kB/s)
    Selecting previously unselected package openssl.

The debconf error is in displayed in red.  It appears twice.

kennethjiang, 2020-07-19:
I'm actually not sure what server 169.44.201.108 is. @RaymondHimle may have a better idea here.

RaymondHimle, 2020-07-19:
This was a host that had the opencv package compiled. Looks like it has
been taken down since then. To get that file you will need to use the
jetson sdk

timur-tabi, 2020-07-19:
How do I get that file via the Jetson SDK?  Like I said, I don't know anything about docker.  How do I tell the installer to use the SDK instead of whatever it's doing now?

I'm also surprised to be the first to notice this.

RaymondHimle, 2020-07-20:
Current workaround is remove the 'dockerfile: Dockerfile.aarch64' from the compose file. And edit the ml_api/Dockerfile line 1 to say 'FROM raymondh2/ml_api:jetson'

robi889, 2020-08-06:
I have the same issue. Is there a solution for this? Im not very good at these things with linux and was following this guide:
https://github.com/MidnightLink/TSDJetsonNano

And the exact same issue happens. 

@RaymondHimle I dont really understand where i have to remove the 'dockerfile: Dockerfile.aarch64' from the compose file. Is there a guide i can follow to achieve this?

timur-tabi, 2020-08-06:
I've had to RMA my Nano, so it'll be at least a week before I can continue working on this.  I will provide updated instructions and files that should make this easier to others to follow.

kennethjiang, 2020-08-07:
Thank you so much @timur-tabi  for offering to help us here!

@robi889 : What @RaymondHimle meant is if you followed the Jetson guide here: https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docs/jetson_guide.md , you'd end up with dockerfile: Dockerfile.aarch64'  in your compose file.

Polymira, 2020-08-09:
> Thank you so much @timur-tabi for offering to help us here!
>
> @robi889 : What @RaymondHimle meant is if you followed the Jetson guide here: [...]

Thanks, this worked for me!

robi889, 2020-08-10:
Thank you @timur-tabi

@kennethjiang Thanks for your explanation. Sadly i dont have the skills to go trough it myself. I didnt follow the official guide, so i will wait if @timur-tabi can figure out a way.

The guide from MidnightLink works fine up to the point where the jetson have to build the container.

thebiggestgit, 2020-10-03:
I had same issue and manually downloaded the file from NVIDIA site (needed a login) then just copied to one of my web servers and changed the URL in the Dockerfile.aarch64 file to download the file from there instead. Would be better to copy locally I guess, but a quick fix for me.

closed 2020-10-04 by raymondh2