TheSpaghettiDetective/obico-server #254: Detective hours at 0 in the docker container, and detective hours page 404'ing

revilo951, 2020-07-21:
Running Spaghetti Detective in docker on an x86 machine. Doesn't seem to be doing any detecting, and the hourglass is showing 0. Clicking on the hourglass tries to load https://mydomain.com//ent/subscription/#detective-hour-balance which gives

> Not Found
> The requested resource was not found on this server.

Docker logs show:
> web_1     | django.request WARNING  Not Found: /ent/subscription/
> web_1     | django.channels.server WARNING  HTTP GET /ent/subscription/ 404 [0.06, 172.30.0.1:52350]

RaymondHimle, 2020-07-21:
This is expected that the hour glass shows zero hours on a private server.

revilo951, 2020-07-21:
OK, thats good to know (perhaps that could be put into the documentation somewhere?).

I figured something might be wrong with my install, because when I turn on the slider for "watch for failures" and then navigate to another page (eg the timelapse page), when I come back to the printer page, the slider is back off. Is this normal?

kennethjiang, 2020-07-23:
@revilo951 just pushed a commit that should remove the DH badge from private server. You can get the latest code by upgrading your server .

commit edaeda9c (2020-07-23): Closes #254. Remove Detective Hours-related UI element from public repo
closed 2020-07-23 by kennethjiang