TheSpaghettiDetective/obico-server #257: Focused Feedback Snapshot Images are Inverted (upside-down) - Not syncing with Octoprint Webcam Settings to Invert Camera

pssnyder, 2020-07-24:
Whenever I access the Focused Feedback to review images, they are loading upside down. When my timelapses are captured and the video feed is streaming "live" the camera is oriented correctly, however, the snapshots are inverted. I should mention that I do have my camera settings modified within OctoPrint to invert my camera because I do have it mounted upside down. My streams and timelapses are oriented correctly in OctoPrint because of those settings modifications from the Webcam Settings screen in OctoPrint. I'm assuming this is just a configuration issue when TSD pulls the video feed. It seems to be using the OctoPrint settings for streaming and for the timelapse video however for snapshots it is ignoring any user adjusted camera settings in OctoPrint.

[image]

kennethjiang, 2020-07-25:
Good catch. This is definitely something we overlooked.

kennethjiang, 2020-08-01:
@pssnyder This fix will be pushed to production tomorrow.

closed 2020-08-01 by kennethjiang

dweenz, 2023-04-24:
was the fix implemented I am getting this also. 04/24/2023

kennethjiang, 2023-05-02:
@dweenz It sounds like a different issue. Please seek help in our discord server https://obico.io/discord/

commit a731cb04 (2020-08-01): Closes #257.