TheSpaghettiDetective/obico-server #259: Can't access webcam from external network

nzzane, 2020-07-25:
As title suggests.

I have TSD running behind a reverse proxy with the domain being on https
TSD and OctoPi are running on the same network, OctoPi is forwarded to its own domain
For some reason TSD keeps trying to access the local octopi IP even when not on the same network, and I am unsure where to change this.
Am i missing something, or is there a setting I need to change in Django

website console shows the following.
docker logs don't show anything out of normal, so seems it is related to the web api?

`VM16:1 Mixed Content: The page at 'https://sub.website.nz/prints/' was loaded over HTTPS, but requested an insecure XMLHttpRequest endpoint 'http://192.168.1.181:3334/media/tsd-timelapses/private/2_p.json'. This request has been blocked; the content must be served over HTTPS.
(anonymous) @ VM16:1
VM16:1 Mixed Content: The page at 'https://sub.website.nz/prints/' was loaded over HTTPS, but requested an insecure XMLHttpRequest endpoint 'http://192.168.1.181:3334/media/tsd-timelapses/private/1_p.json'. This request has been blocked; the content must be served over HTTPS.
(anonymous) @ VM16:1
2createError.js:16 Uncaught (in promise) Error: Network Error
    at t.exports (createError.js:16)
    at XMLHttpRequest.h.onerror (xhr.js:83)
PrintsPage.vue:203 Object
VM16:1 Mixed Content: The page at 'https://sub.website.nz/prints/' was loaded over HTTPS, but requested an insecure XMLHttpRequest endpoint 'http://192.168.1.181:3334/media/tsd-timelapses/private/2_tagged.mp4'. This request has been blocked; the content must be served over HTTPS.
(anonymous) @ VM16:1
DevTools failed to load SourceMap: Could not load content for https://sub.website.nz/static/vendor/js/filesize.min.js.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE`

RaymondHimle, 2020-07-25:
Please make sure your Django site is setup with the correct url.

nzzane, 2020-07-25:
> Please make sure your Django site is setup with the correct url.

Just double checked, it is setup with the correct URL, anything else I may be missing?

RaymondHimle, 2020-07-26:
It should be the url you intend to use. So if you are using a domain name you should be using that

nzzane, 2020-07-26:
> It should be the url you intend to use. So if you are using a domain name you should be using that

Okay so changed to the full url (no https, and no longer the ip) and webcam is working, however timelapse is still trying to get the local IP

RaymondHimle, 2020-07-26:
This is likely related to https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/issues/243

RaymondHimle, 2020-07-26:
Is your server up to date? I cannot reproduce this issue on my end.

nzzane, 2020-07-26:
> Is your server up to date? I cannot reproduce this issue on my end.

Yeah, just confirmed with a git pull, no changes.  Remade with docker-compose just to be sure.
Running a new print now to see how it goes

kennethjiang, 2020-07-26:
Just pushed a commit to better document reverse proxy: https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/commit/5adb81d33ec384407ded7986abc517bb07fea619

Can you take a look at https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docs/advanced_config.md#using-a-reverse-proxy to see if you are still missing something @nzzane ? If not, please post the error messages you are getting.

nzzane, 2020-07-26:
Figured it out
Time lapses from before I changed site in django were causing the above error in browser, deleted these and did a new print using the FQDN in django and now the timelapses load fine, and can download /view them remotely 

Thanks for the help guys!

closed 2020-07-26 by nzzane