TheSpaghettiDetective/obico-server #260: Web and Tasks containers keep crashing

serenewaffles, 2020-07-29:
I installed Ubuntu server 20.04 LTS today on a fresh hard drive. The first thing I did was install docker-compose and attempt to build The Spaghetti Detective. The Web and Tasks containers are constantly crashing and restarting for me. It seems similar to #194 and #207. I am using `sudo docker-compose up --build -d` to launch TSD per the manual. 

```
lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.1 LTS
Release:        20.04
Codename:       focal
```
```
sudo docker version
Client:
 Version:           19.03.8
 API version:       1.40
 Go version:        go1.13.8
 Git commit:        afacb8b7f0
 Built:             Tue Jun 23 22:26:12 2020
 OS/Arch:           linux/amd64
 Experimental:      false

Server:
 Engine:
  Version:          19.03.11
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.13.12
  Git commit:       77e06fd
  Built:            Mon Jun  8 20:24:59 2020
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          v1.2.13
  GitCommit:        7ad184331fa3e55e52b890ea95e65ba581ae3429
 runc:
  Version:          1.0.0-rc10
  GitCommit:
 docker-init:
  Version:          0.18.0
  GitCommit:        fec3683
```

```
sudo docker-compose version
docker-compose version 1.25.0, build unknown
docker-py version: 4.1.0
CPython version: 3.8.2
OpenSSL version: OpenSSL 1.1.1f  31 Mar 2020
```

Logs for the Web container:
```
Traceback (most recent call last):
  File "manage.py", line 15, in <module>
    execute_from_command_line(sys.argv)
  File "/usr/local/lib/python3.6/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line
    utility.execute()
  File "/usr/local/lib/python3.6/site-packages/django/core/management/__init__.py", line 325, in execute
    settings.INSTALLED_APPS
  File "/usr/local/lib/python3.6/site-packages/django/conf/__init__.py", line 79, in __getattr__
    self._setup(name)
  File "/usr/local/lib/python3.6/site-packages/django/conf/__init__.py", line 66, in _setup
    self._wrapped = Settings(settings_module)
  File "/usr/local/lib/python3.6/site-packages/django/conf/__init__.py", line 157, in __init__
    mod = importlib.import_module(self.SETTINGS_MODULE)
  File "/usr/local/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 941, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
[... 581 lines cut ...]
    import raven
  File "/usr/local/lib/python3.6/site-packages/raven/__init__.py", line 49, in <module>
    from raven.base import *  # NOQA
  File "/usr/local/lib/python3.6/site-packages/raven/base.py", line 37, in <module>
    from raven.conf.remote import RemoteConfig
  File "/usr/local/lib/python3.6/site-packages/raven/conf/remote.py", line 36, in <module>
    DEFAULT_TRANSPORT = discover_default_transport()
  File "/usr/local/lib/python3.6/site-packages/raven/conf/remote.py", line 18, in discover_default_transport
    from raven.transport.threaded import ThreadedHTTPTransport
  File "/usr/local/lib/python3.6/site-packages/raven/transport/__init__.py", line 18, in <module>
    from raven.transport.registry import *  # NOQA
  File "/usr/local/lib/python3.6/site-packages/raven/transport/registry.py", line 18, in <module>
    from raven.transport.twisted import TwistedHTTPTransport
  File "/usr/local/lib/python3.6/site-packages/raven/transport/twisted.py", line 16, in <module>
    from twisted.web.client import (
  File "/usr/local/lib/python3.6/site-packages/twisted/web/client.py", line 28, in <module>
    from zope.interface import implementer
  File "/usr/local/lib/python3.6/site-packages/zope/interface/__init__.py", line 53, in <module>
    from zope.interface.interface import Interface
  File "/usr/local/lib/python3.6/site-packages/zope/interface/interface.py", line 654, in <module>
    {'__slots__': ()}
  File "/usr/local/lib/python3.6/site-packages/zope/interface/interface.py", line 629, in __new__
    moduledescr = InterfaceBase.__dict__['__module__']
KeyError: '__module__'
```

kennethjiang, 2020-07-29:
It looks like we introduced some breaking changes at some point. investigating now. stay tuned.

kennethjiang, 2020-07-29:
@serenewaffles Can you pull the latest commits and try it again?

closed 2020-07-29 by kennethjiang

serenewaffles, 2020-07-29:
I did another `git pull` followed by `docker-compose up --build -d` and it is working now.

Thank you!

commit 885b0be8 (2020-07-29): Closes #260. For some mysterious reason zope.interface 5.1.0 is breaking thing although it was released in April