TheSpaghettiDetective/obico-server #264: docker web container: ValueError: Couldn't load manifest 'staticfiles.json' (version 1.0)

chris948, 2020-08-07:
Sorry I can't be more help, I can't get into the container because it won't load, and I tried to revert to previous versions, I suspect the prior to python upgrade a few days ago would work, but when I build it, it just pulls the latest from here.

Ubuntu 20.04, docker, using docker-compose up --build -d

`Traceback (most recent call last):
  File "manage.py", line 15, in <module>
    execute_from_command_line(sys.argv)
  File "/usr/local/lib/python3.7/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line
    utility.execute()
  File "/usr/local/lib/python3.7/site-packages/django/core/management/__init__.py", line 375, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/usr/local/lib/python3.7/site-packages/django/core/management/base.py", line 323, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/usr/local/lib/python3.7/site-packages/django/core/management/base.py", line 364, in execute
    output = self.handle(*args, **options)
  File "/usr/local/lib/python3.7/site-packages/django/contrib/staticfiles/management/commands/collectstatic.py", line 162, in handle
    if self.is_local_storage() and self.storage.location:
  File "/usr/local/lib/python3.7/site-packages/django/contrib/staticfiles/management/commands/collectstatic.py", line 216, in is_local_storage
    return isinstance(self.storage, FileSystemStorage)
  File "/usr/local/lib/python3.7/site-packages/django/utils/functional.py", line 256, in inner
    self._setup()
  File "/usr/local/lib/python3.7/site-packages/django/contrib/staticfiles/storage.py", line 498, in _setup
    self._wrapped = get_storage_class(settings.STATICFILES_STORAGE)()
  File "/usr/local/lib/python3.7/site-packages/django/contrib/staticfiles/storage.py", line 377, in __init__
    self.hashed_files = self.load_manifest()
  File "/usr/local/lib/python3.7/site-packages/django/contrib/staticfiles/storage.py", line 399, in load_manifest
    (self.manifest_name, self.manifest_version))
ValueError: Couldn't load manifest 'staticfiles.json' (version 1.0)`

kennethjiang, 2020-08-07:
What OS are you running for your host?

Also can you reclone the repo to a new folder and start over?

Also if you still have errors in the newly cloned repo, you can run `docker-compose up --build | tee compose.log`, then attach the `compose.log` file here.

chris948, 2020-08-08:
I don't get it, I repulled twice without the extra logging, and it didn't work. I repulled with the logging and it took 10 minutes, but whatever it built actually works. I know that doesn't make sense, but thank you.

Thanks

closed 2020-08-08 by chris948