TheSpaghettiDetective/obico-server #265: Apache forwarding configuration

corporategoth, 2020-08-07:
FYI,

Since you don't have it in your docs, here is the forwarding configuration for Apache.  I specifically wanted the URL to be /tsd, as opposed to the root of the webserver - but your hard-coding of URLs without a definable prefix made this difficult:

Using a website (in the admin interface) configured as <mysite>/tsd, the following configuration is working for me:

```
    ProxyRequests off

    RewriteEngine on
    RewriteRule ^/tsd/ws(.*) ws://localhost:3334/ws$1 [P,L]

    ProxyPass /ws/ ws://localhost:3334/ws/
    ProxyPassReverse /ws/ ws://localhost:3334/ws/

    ProxyPass /tsd/ http://localhost:3334/
    ProxyHTMLURLMap http://localhost:3334 /tsd

    <Location /tsd/>
        ProxyPassReverse /
        ProxyHTMLEnable on
        ProxyHTMLExtended on
        ProxyHTMLURLMap \"/([^/].*) \"/tsd/$1 Rhe
        ProxyHTMLURLMap ^/([^/].*) /tsd/$1 Rc
        RequestHeader    unset  Accept-Encoding
    </Location>
```

Unfortunately it's not super clean, having to do a URL rewrite from /tsd/ws/<foo> -> /ws/<foo> before the proxypass will work for websockets.  But for now, it's working.

Not a bug, but figured you might like to put it in your docs.

kennethjiang, 2020-08-07:
Thank you so much for bringing this to us!

Are you interested in opening a PR for making the root URL configurable? I remember a couple other users have brought this up and I agree it'll make the Apache config file a lot cleaner.

corporategoth, 2020-08-07:
Updated my coinfig

re: the PR, that is a lot of work - mainly due to you using it everywhere across js and html files.
Since you're not using a dynamic language (like php, perl, python, or etc) to generate pages, but are using static js/html, the only way for you to add a prefix would be to ALWAYS use JS to generate a url, as HTML doesn't have a variable syntax.

You would literally need to change all <a> records to some kind of javascript.

Plus, it seems there are different combinations of libraries, for example, I still could not get the time lapse page to render correctly, because it references /api/v1/prints which is defined in frontend/src/lib/apis.js from what I can tell, and no attempts I have made at replacement so far are doing anything.

It would be awesome to not have to do this, but I just don't have time to dig into everywhere you are using fixed URLs right now.

kennethjiang, 2020-08-07:
Hmmm, not sure what you mean by we are using static js/html. if you look closely, we are using Django (believed to be the most common Python-based framework).