TheSpaghettiDetective/obico-server #271: Secret Token don't show up when adding new printer

Lyan9145, 2020-08-18:
like this
[image]
Does it generate locally or some where else?
or it downloads the script on the internet
My network has some problems and some server is probably not reachable

And the "advanced settings" button also not responding when I clicked it

kennethjiang, 2020-08-18:
I'm not sure I'm following. Have you followed this guide: https://www.thespaghettidetective.com/docs/octoprint-plugin-setup/ ? if so, on which step did you get stuck or see an error?

Lyan9145, 2020-08-19:
Sorry I didn't state the problem clearly.
I build a private server, when adding new printer, the secret token doesn't show up and the "copy" button doesn't work

kennethjiang, 2020-08-19:
Can you post the relevant screenshots for the problems you are getting?

Lyan9145, 2020-08-19:
[image]

kennethjiang, 2020-08-19:
Where did you get that screenshot from? The new printer page looks like this:

[image]

Lyan9145, 2020-08-19:
So did I build the version of the server? 
I just followed the guide on the github page :(

kennethjiang, 2020-08-19:
I hate to tell you do more work but I'm still quite confused about this issue. But can you please provide the step-by-step description to reproduce this error?

Lyan9145, 2020-08-21:
First git clone `https://github.com/TheSpaghettiDetective/TheSpaghettiDetective.git`
Then I delete the `RUN wget --quiet -O model/model.weights $(cat model/model.weights.url | tr -d '\r')` in ml_api/dockerfile and manually download the model.weight file and put it in ml_api/model folder.(Because the wget is too slow)
[image]

[image]

Last `docker-compose up -d`
It will run into some network issue like time out or some thing else, retry it for a couple of time and it will success

To configure the server I go to `http://192.168.x.x:3334/admin/`
and change "domain name" to 192.168.x.x:3334

and...
still the same...
I think it most likely to be my environment's issue,

I investigate the `http://192.168.x.x:3334/printers/new/` page source code and found
`<script src="https://code.jquery.com/jquery-3.4.1.min.js" integrity="sha256-CSXorXvZcTkaix6Yvo6HppcZGetbYMGWSFlBw8HfCJo=" crossorigin="anonymous"></script>`
this javascript(line 311) is unreachable
[image]

EuphoricPenguin, 2020-08-24:
Couldn't this issue be solved by hosting jQuery locally?

kennethjiang, 2020-08-24:
In theory we can host jquery on TSD server. but then we may have to host all other js, css, sourcemap etc, as any of them may fail to load from a browser that doesn't have a solid internet connection. Not only will it add more load to our TSD cloud server, but it also slow down the page loading unless we set up a very solid CDN.

My suggestion will be to figure out why the connection to code.jquery.com is not working and fix it.

Lyan9145, 2020-08-24:
I think there is probably a firewall or something that the internet provider have which is blocking my browser. I don't think they would solve the problem for me.

I tried adding printer manually in the Django admin page, the octoprint plugin can be connected to the server, but it don't shown up in the printer page. I will try to print something tomorrow and see if it works.
[image]
[image]
[image]

tazitoo, 2020-11-09:
I had a similar issue - the whole page rendering at once (...no 'Next' buttons...) and nothing populated in the token area.

In firefox, I run several privacy extensions (e.g. ghostery, privacy possum, cookie autodelete, and localcdn).  Things worked as expected when I temporarily disabled them.  Hoping this comment helps others if they run into similar issues.

closed 2022-04-26 by raymondh2