TheSpaghettiDetective/obico-server #275: Suggestion: webcam streaming for the mobile app

goeland86, 2020-08-30:
As discussed with Kenneth on discord, I had a suggestion regarding the use of the mobile app with private servers.

Instead of trying to use a webRTC setup where the app gets the stream from the TSD server, allow the user to manually input the camera stream URL directly in the app, like OctoPrint does?

Obviously the highlight of potential spaghetti detected would be missing - but I don't think it's something that would be particularly required.

kennethjiang, 2020-08-30:
Thank you @goeland86 for opening this issue. A follow-up question: how would the app decide if it should stream from WebRTC or the manually-entered URL?

goeland86, 2020-08-30:
I would aim for a fail-over strategy.

So if the manually entered url is accessible (i.e not throwing a 404), display that, if it isn't, fallback to web RTC?

For private server instances there may be no fallback, and that's something people will have to cope with in the short term.

Others may have different needs, but I envision based on my use case where I'm streaming the camera feed to the web anyway.

kennethjiang, 2020-08-30:
Hmmm, what if the url was initially available, but then users switched to a different network? Should the app continuously polls the url? It feels a bit hacky to me as it reminds me of similar workarounds in other projects that mostly ended up opening a can of worms...

Can we take a step back to look at the problem we are trying to solve here? Is it because you run into problem with WebRTC streaming for your private server?

goeland86, 2020-08-30:
In my case the private server is hidden behind a firewall in the workshop, and not publicly accessible. The webcam streams are available through port forwarding though.

For me the webrtc is always at a limited fps, while the camera feed is real-time and higher resolution (720p). If I'm trying to diagnose if there's a problem from the detection, the higher resolution would be better.

kennethjiang, 2020-08-30:
Ah sorry I didn't fully understand your situation in the discord conversation.

webrtc uses hole-punching that can go through most firewalls except the most stringent ones. Is it possible to configure your filrewall to allow that? Just google "firewall configuration webrtc" and you should see that the requirements are not very crazy.

goeland86, 2020-08-30:
The firewall should allow it - it's nothing too stringent. But for me it's that the webRTC refresh rate is exceedingly slow. Like 1 frame per second.

kennethjiang, 2020-08-30:
I assume you have gone through the trouble-shooting steps here? https://www.thespaghettidetective.com/docs/webcam-feed-is-laggy/

goeland86, 2020-08-30:
I didn't find the page, but yes - all the various steps were tried.

kennethjiang, 2020-08-30:
Ok. I'd view this as an issue about maximizing the chance for webrtc to work, instead of putting on a band-aid to mjpeg streaming. The issues that popped in my mind include:
- The fallback mechanism as discussed earlier. Continuous polling which is harsh on battery.
- Webrtc works for most users (95% or more including private servers). So using the inefficient mjpeg streaming by default doesn't make sense.
- mjpeg streaming itself is already harsh on battery even without polling.
- by default OctoPrint mjpeg streaming is not authenticated. So the possibility of blindly connecting to a different webcam streaming and causing privacy issues.

And my gut tells me we would run into more unexpected issues should we proceed with this approach. The rarity of this problem doesn't seem to justify the complexity and intricacy of the solution...

But it's possible that I missed something that could make this approach simple and robust. I'm open to further discussion.

Sorry again that I misunderstood the problem initially.

goeland86, 2020-08-31:
Not a problem. I also got confused when there was mention of licensing issues - I thought this might be a neat way to side-step it.

closed 2020-08-31 by goeland86

kennethjiang, 2020-08-31:
Ah sorry I was not clear. The licensing issue is only around TURN server, which is usually not required to do webrtc.