TheSpaghettiDetective/obico-server #280: Create Mainsail/Moonraker plugin

RaymondHimle, 2020-09-05:
https://github.com/Arksine/moonraker
https://github.com/meteyou/mainsail

stefanpieter, 2020-09-20:
Yes please!

CraXyOW3, 2021-01-29:
This is the only reason i am not going over to fluidd/moonraker! Please make it so!

LazaroFilm, 2021-03-10:
Going to add my voice to this. It would be great to have the ability to use it without Octoprint.

gmergulhao, 2021-03-18:
+1 on that!

readeral, 2021-03-25:
Also very keen - I’ve just started using Klipper and am a Pro subscriber for TSD but am thinking of cancelling simply because I like Moonraker too much

ashleyuk90, 2021-03-27:
Also a + from me, or a way to at least run TSD manually stand alone without it needing to be fed when a "print" is running so can still be used to send alerts of suspicious objects.

necropolis04, 2021-03-29:
Yes please, i would need it for fluidd

ale624, 2021-04-08:
Yep! Would also absolutely love this. Octoprint has such a bloat factor compared with Fluidd!

Nickperillo, 2021-04-24:
Would love to see this implemented!

intrepidsilence, 2021-05-17:
Yes, please. +1

IamGaryGnu, 2021-05-26:
I'd love this too, pretty please?  
I'm on Fluidd now, if I can't get TSD on Klipper i'll just have my subscription run out and not renew, but I love this bad boy and would far prefer to keep it!

kennethjiang, 2021-05-27:
This issue is currently on hold as it is trickier than we initially thought. We will resume it once we have wrapped up other high-priority features we are working on right now. Sorry... :(

meefischli, 2021-06-06:
I am on Fluidd/Klipper too now, subscription will end too. Hope you find a solution.

valinhorn, 2021-06-25:
I have been checking out Fluidd/Klipper and TSD is the only thing keeping me with Octoprint. I would love to see this feature added. I have been a big supporter and I will continue to do so but I would love the option to use Fluidd/Mainsail instead of Octoprint.

kennethjiang, 2021-06-25:
Thank you everyone for adding voice to this issue. You have made it clear that this is the next big thing we should focus on. This is a "big thing" because Moonraker doesn't have a plugin structure as OctoPrint does and hence requires a significant amount of effort. However, we can see the wave of Klipper is sweeping every corner of the 3D printing community. We will be happy to ride with it instead of fight it. :)

LazaroFilm, 2021-07-05:
Amazing news! Looking forward to seeing this happen !

mjg2688, 2021-07-05:
Just made the change over to Kilpper/Mainsail myself, also excited to see the addition of plug-in support.  Bummed I lost TSD, but the trade off is worth it in my opinion.  Great to see how active development is here!

rullywowpcb, 2021-08-30:
So glad to hear you are considering it. Just switched to Klipper/Fluidd and am loving everything about it. Thanks!

tomlawesome, 2021-08-31:
+1 for Klipper and Mainsail 

Looking forward to seeing it, and using it!

clomads, 2021-09-06:
+1 for Mainsail/Moonraker integration

Jo3yDaPh1sH, 2021-09-08:
Thank you very much for considering making this plugin. I would love to use it too.. it's definitely the missing link to complete My experience with klipper and the beauty of mainsail.  Either way excellent work.

crashmaxx, 2021-09-10:
I would be happy to see a basic plugin with minimal functionality. Just link the webcam stream to the Spaghetti Detective server and do the analysis. That would be enough to get failure notifications.

tomlawesome, 2021-09-13:
> I would be happy to see a basic plugin with minimal functionality. Just link the webcam stream to the Spaghetti Detective server and do the analysis. That would be enough to get failure notifications.

Me too, what I would actually like to do is run TSD on my server with a webcam stream linked to it, just so that I can get remote alerts via email/telegram/text that way it doesn’t require any integration.

simon021, 2021-09-15:
Would love to see this added.  Basic functionally would be totally fine.  As suggested above just webcame feed out for analysis would be plenty to get us started!

Thanks!

crimsonthunder, 2021-09-29:
Thank you for considering the implementation! I look forward to it!

kennethjiang, 2021-10-14:
It's being worked on. Stay tuned.

Zayonx-0, 2021-11-15:
Looking forward to this ! Really happy that you guys are working on it ;)

kennethjiang, 2021-11-15:
We have something that's quite raw. But if you feel adventurous, feel free to take a swing at it: https://github.com/TheSpaghettiDetective/tsd-moonraker

hkr1990, 2021-11-25:
Would love to use the detective with Klipper. Please prioritize it developers.

ge123, 2021-12-13:
Hello, I tried installing the alpha, but when I try git pull I get fatal: not a git repository (or any of the parent directories): .git. What am i doing wrong?

crashmaxx, 2021-12-13:
@ge123 

Here is how I was able to install it.

# requires python3; install python3 packages
sudo apt-get install python3 python3-pip python3-venv

# clone repo
git clone https://github.com/TheSpaghettiDetective/tsd-moonraker.git

# setup virtual environment
python3 -m venv tsd-moonraker/
cd tsd-moonraker
source venv/bin/activate
pip3 install -r requirements.txt

# fill in essential configuration
cp config.sample.ini config.ini
nano config.ini

# link printer (grab tsd auth token)
python3 -m tsd_moonraker.link -c config.ini

# start app
python3 -m tsd_moonraker.app -c config.ini -l tsd_moonraker.log

ge123, 2021-12-13:
Thank you!  I’ll give it a shot 😀.

kennethjiang, 2021-12-13:
It seems some users have had success with the moonraker plugin  https://github.com/TheSpaghettiDetective/tsd-moonraker.git

I'm closing this issue. For the issues related to the plugin, let's file it in its own repo.

closed 2021-12-13 by kennethjiang