TheSpaghettiDetective/obico-server #292: "Service 'web' failed to build" on Jetson Nano

amolloy, 2020-09-16:
Following the [Jetson Nano guide](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docs/jetson_guide.md), building the web component fails with the error:

`Building web
Step 1/7 : FROM thespaghettidetective/web:base-1.1
base-1.1: Pulling from thespaghettidetective/web
ERROR: Service 'web' failed to build: no matching manifest for linux/arm64/v8 in the manifest list entries`

I've attached the complete log for reference. 
[JetsonNanoInstallLog.txt](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/files/5232021/JetsonNanoInstallLog.txt)

This is my first time working with docker-compose, so hopefully I've just missing some step. For docker-container.override.yml, I have:

`-> % cat docker-compose.override.yml 
version: '2.4'

x-web-defaults: &web-defaults
  build:
    dockerfile: Dockerfile.base

services:
  ml_api:
    build:
      context: ml_api
      dockerfile: Dockerfile.aarch64
    environment:
        HAS_GPU: 'True'
    runtime: nvidia
`

Thanks for the help!

kennethjiang, 2020-09-17:
@RaymondHimle Is it because I messed up something when creating the manifest?

amolloy, 2020-09-30:
Hi, sorry, I don't mean to be a pest, just wondering if there's been any movement on this? I'd love to get TSD running on my Nano, but I haven't got enough understanding of how it all fits together to be much good at solving it myself. Thanks so much!

kennethjiang, 2020-09-30:
Sorry for the delay. Apparently I tagged the wrong handle...

@raymondh2 just updated the Jetson guide. @amolloy please follow the updated document at: https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/docs/jetson_guide.md ?

amolloy, 2020-09-30:
Thanks! That did the trick, I've got it working now. I appreciate the help!

closed 2020-09-30 by amolloy