TheSpaghettiDetective/obico-server #296: Implement 2-Factor Authentication

as-kholin, 2020-09-21:
For various (good) reasons, Octoprint is recommended to not be made available directly on the Internet (unless compensating controls are added).  Especially with tunneling, if someones TSD account gets hacked, then they have effectively done just that.  Not saying that TSD does not add some security, only that people being people, chances are this will happen to at least some due to poor password management procedures.

While we cannot prevent that entirely, we can make it materially harder to occur by allowing (or possibly, in the case of tunneling, requiring) 2-Factor Authentication/Multi-factor Authentication.

Request enhancement to add at least TOTP (Google Authenticator, Twilio Authy, etc), possibly including some other 2FA mechanisms (U2F, FIDO2, Yubikey, etc).  Defer to others if any sort of SMS auth is worth considering, given criticisms of this route vs. ubiquity.

kennethjiang, 2020-09-21:
It is a very good point. And a PR will be really appreciated! :)

donicrosby, 2020-12-12:
There's already a Django library https://github.com/Bouke/django-two-factor-auth (MIT Licensed) that you could use that has everything that you would want for this issue. I can implement this and make a PR if it's acceptable to use this 3rd party library.

kennethjiang, 2020-12-12:
Of course we can use 3rd party lib. We are using a ton of them. :)

Yes a PR will be very welcome!

GJSchaller, 2024-01-07:
Checking back on this, is there any way a non-technical person can help?

Cthulhu709, 2025-02-10:
Please get this implemented.  With the exploitation of reused credentials rampant, and the link of our personal home and business printers connected to this service, a hack has a potentially dangerous outcome in the physical sense. I'll not be renewing when my subscription comes due shortly if this issue is not fixed.

Cthulhu709, 2025-02-10:
> Request enhancement to add at least TOTP (Google Authenticator, Twilio Authy, etc), possibly including some other 2FA mechanisms (U2F, FIDO2, Yubikey, etc). Defer to others if any sort of SMS auth is worth considering, given criticisms of this route vs. ubiquity.

From a cybersecurity practitioner's perspective, it's a good/better/best scenario and who you want to exclude.

Google Authenticator is the universally accepted generator, but that same QR can be used on most other code generator (PingID/Authy/etc.) applications.  While Google has it's own security issues, it's one that I do put a large part of my trust in.

Email/SMS should likely be excluded at this point due to their likelihood of exploitation.  Email accounts are often hijacked, and that can grant extensive privilege to many other services.  But unfortunately, if your userbase contains older (senior) individuals who have a hard time with better MFA services, it could limit your customer base size somewhat.

thx1200, 2025-04-22:
I'm very interested in this as well.  I wish I could create the PR myself but I don't know anything about how django actually works.