TheSpaghettiDetective/obico-server #302: Telegram Login not visible

danyial, 2020-10-15:
TSD local server:

It's not possible to log in to Telegram.

[image]

kennethjiang, 2020-10-15:
I remember going through a very delicate dance to set up telegram bot in TSD cloud servers. Will really appreciate if anyone can figure out a repeatable process for private servers and document it.

t413, 2020-10-27:
you've definitely got your `x-web-defaults: environment: TELEGRAM_BOT_TOKEN:` set in your docker-compose.yml? Should look something like 
```
x-web-defaults: 
  ..
  environment: 
    ..
    TELEGRAM_BOT_TOKEN: 3839102389:FJDfaLsfdaA-EJF44vdsvGHDsgFSDgfdSvf
```

kennethjiang, 2020-10-27:
@t413 I redacted your token as I believe it should be kept secret.

t413, 2020-10-27:
Hah, definitely should. That's why I keyboard-babled that one to match the key length. 😅

danyial, 2020-11-05:
Thats my docker-compose.yml. I'm pretty sure that everything is fine with my config 🤓

`version: '2.4'

x-web-defaults: &web-defaults
  restart: unless-stopped
  build:
    context: web
    dockerfile: "Dockerfile${DOCKERFILE_VERSION}"
  volumes:
    - ./web:/app
  depends_on:
    - redis
  environment:
    EMAIL_HOST:     # -> such as 'smtp.gmail.com'
    EMAIL_HOST_USER:   # -> such as your email address for a Gmail account
    EMAIL_HOST_PASSWORD:    # -> your email account password
    EMAIL_PORT: 587  # DO NOT surround it with quotes. Otherwise email won't be sent!
    EMAIL_USE_TLS: 'True'
    DEFAULT_FROM_EMAIL: 'changeme@example.com'
    DEBUG: 'False'    # Don't set DEBUG to True, otherwise the static files will be cached in browser until hard-refresh
    SITE_USES_HTTPS: 'False' # set it to 'True' if https is set up
    SOCIAL_LOGIN: 'False'
    REDIS_URL: 'redis://redis:6379'
    DATABASE_URL: 'sqlite:////app/db.sqlite3'
    INTERNAL_MEDIA_HOST: 'http://web:3334'
    ML_API_HOST: 'http://ml_api:3333'
    ACCOUNT_ALLOW_SIGN_UP: 'False'  # -> set to 'True' if you want to open sign up form
    WEBPACK_LOADER_ENABLED:
    TELEGRAM_BOT_TOKEN: '0000000000:11111111111111111111111111111111111'

    #### Optional env vars below this line ###
    #TWILIO_ACCOUNT_SID: https://django-twilio.readthedocs.io/en/latest/settings.html for how to find and set the TWILIO_XXX vars
    #TWILIO_AUTH_TOKEN:
    #TWILIO_FROM_NUMBER:
    #ENALBE_HISTORY:
    #ML_API_TOKEN:
    #SENTRY_DSN:
    #TELEGRAM_BOT_TOKEN: https://core.telegram.org/bots#6-botfather for how to create a telegram bot and get its token

services:
  ml_api:
    hostname: ml_api
    restart: unless-stopped
    build:
      context: ml_api
    environment:
        DEBUG: 'True'
        FLASK_APP: 'server.py'
        #ML_API_TOKEN:
        #HAS_GPU: 'False'
    command: bash -c "gunicorn --bind 0.0.0.0:3333 --workers 1 wsgi"

  web:
    <<: *web-defaults
    hostname: web
    ports:
      - "3334:3334"
    depends_on:
      - ml_api
    command: sh -c "python manage.py collectstatic --noinput && python manage.py migrate && python manage.py runserver --noreload 0.0.0.0:3334"

  tasks:
    <<: *web-defaults
    hostname: tasks
    command: sh -c "celery -A config worker --beat -l info -c 2 -Q realtime,celery"

  redis:
    restart: unless-stopped
    image: redis:5.0-alpine
`

LuxuSam, 2020-12-11:
I'm having the same issue with telegram. 

I get this window in **chrome**:
[image]
And this in **safari**:
[image]

I got the following error in the console when I try to reload the frame in chrome with "reload frame":
`Refused to frame 'https://oauth.telegram.org/' because an ancestor violates the following Content Security Policy directive: "frame-ancestors http://mysite.url"`

LuxuSam, 2020-12-14:
I have got it working by using the suggested https://localtunnel.github.io/www/ instead of my own site url. 
It seems like telegram bots do not like ports.

kennethjiang, 2020-12-14:
Yes now I remember telegram does have a policy to mandate https for a webhook. See their doc here: https://core.telegram.org/bots/webhooks#always-ssl-tls

I'll close this issue as this is not a code issue. Other users who run into the same issue in the future can still find this post using google.

closed 2020-12-14 by kennethjiang