TheSpaghettiDetective/obico-server #306: Error while upgrading server

LuxuSam, 2020-10-29:
Hi, I wanted to upgrade my server today but I got the following error:
```Collecting grpcio<2.0dev,>=1.29.0; extra == "grpc"
  Downloading grpcio-1.33.2.tar.gz (20.9 MB)
    ERROR: Command errored out with exit status 1:
     command: /usr/local/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-wqumflmu/grpcio/setup.py'"'"'; __file__='"'"'/tmp/pip-install-wqumflmu/grpcio/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-84j0koxd
         cwd: /tmp/pip-install-wqumflmu/grpcio/
    Complete output (11 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-wqumflmu/grpcio/setup.py", line 224, in <module>
        if check_linker_need_libatomic():
      File "/tmp/pip-install-wqumflmu/grpcio/setup.py", line 174, in check_linker_need_libatomic
        stderr=PIPE)
      File "/usr/local/lib/python3.7/subprocess.py", line 800, in __init__
        restore_signals, start_new_session)
      File "/usr/local/lib/python3.7/subprocess.py", line 1551, in _execute_child
        raise child_exception_type(errno_num, err_msg, err_filename)
    FileNotFoundError: [Errno 2] No such file or directory: 'c++': 'c++'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
ERROR: Service 'web' failed to build: The command '/bin/sh -c pip install -r requirements.txt' returned a non-zero code: 1
```
The attached file contains the complete buildlog
[buildlog.log](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/files/5458856/buildlog.log)

kennethjiang, 2020-10-31:
`grpcio` should be pre-built in the docker base image. It's quite strange.

Can you run:

```
docker run --rm -it thespaghettidetective/web:base-1.2 bash
pip list  | grep grpcio # run this when the previous command shells you in the container
```

and post the console output here? I suspect you have a corrupted docker image cached locally...

LuxuSam, 2020-10-31:
I got it up running again yesterday with the help of someone from your discord. Should I still do that?

kennethjiang, 2020-10-31:
If it's already working for you, you don't need to do it. In this case I'll just close this issue.

closed 2020-10-31 by kennethjiang

raymondh2, 2020-11-12:
@kennethjiang you may need to update the images on Docker hub to fix this. Looks like it was an issue with the Alpine base image.

kennethjiang, 2020-11-14:
Rebuilt the image and pushed. Please post it here if anyone is still having any issues.