TheSpaghettiDetective/obico-server #307: Feature request: Integration in Repetier-Server

G4Cab, 2020-10-31:
Looking forward for integration in Repetier-Server

https://www.repetier-server.com/

kennethjiang, 2020-10-31:
I actually never understood how repetier-server works. I'm under the impression it's a closed system but I could be wrong.

Do you know if there is some kind of plugin mechanism so that we can run code inside it? If so, some pointers to the docs will be really appreciated.

G4Cab, 2020-10-31:
You have an API
https://www.repetier-server.com/manuals/programming/API/index.html

You can use LUA modules
https://www.repetier-server.com/manuals/programming/API/modules.html
https://www.repetier-server.com/monitoring-plugins/

You also have some server commands
https://www.repetier-server.com/manuals/0.94/index.html

The developer itself is interested in integrating SD, maybe you can get in contact with him:
https://forum.repetier.com/discussion/comment/33224/#Comment_33224

kennethjiang, 2020-10-31:
Thank you so much for all these pointers. I have heard good words about repetier. It looks like they are more sophisticated than I thought. :)

This feels like a pretty big undertaking. And almost the whole TSD team is tied up with mobile app right now. When things settle down a bit we will evaluate the amount of work and prioritize it accordingly.

Thanks again for this suggestion.

dragnmastr85, 2024-02-12:
I would like to second this request!