TheSpaghettiDetective/obico-server #308: API

Spirtogram, 2020-11-03:
Hello!  I have a local server. How can I get the sensor state from it for transmission to the home assistant server.

kennethjiang, 2020-11-03:
We don't currently have a public API for such integration. What you can do right now is to open developer tools in chrome and check the messages in the websocket connections. You will see 2 websocket connections. One of them is for webcam streaming which you can ignore for your purpose.

What home assistant server are you trying to integrate?

kennethjiang, 2020-11-03:
Also I'm closing this issue as the duplicate of #195 . Please provide more info in that issue so that we can start to collect the requirements around a public API.

closed 2020-11-03 by kennethjiang