TheSpaghettiDetective/obico-server #310: Making dataset available

hbrylkowski, 2020-11-08:
Hi,

I would want to try and train my own models, however I couldn't find the dataset that you are using. I've only found link in this issue - https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/issues/21 to google drive - would it be ok if I just request for access to said gdrive?

kennethjiang, 2020-11-09:
Reactivated the [link](https://drive.google.com/drive/folders/1IpGAPbiHYDJFsTtlBEktLGq6niTvB_BJ?usp=sharing). 

Please let me know if you can access it so that I can close this issue.

CoderChen01, 2020-11-17:
> Reactivated the [link](https://drive.google.com/drive/folders/1IpGAPbiHYDJFsTtlBEktLGq6niTvB_BJ?usp=sharing).
> 
> Please let me know if you can access it so that I can close this issue.

Hi, How do I label the target box for this data set, in other words, how to put the target box on the spaghetti to make the model work best, like this?

CoderChen01, 2020-11-17:
> Reactivated the [link](https://drive.google.com/drive/folders/1IpGAPbiHYDJFsTtlBEktLGq6niTvB_BJ?usp=sharing).
> 
> Please let me know if you can access it so that I can close this issue.

[image]

CoderChen01, 2020-11-17:
[image]

CoderChen01, 2020-11-17:
> Reactivated the [link](https://drive.google.com/drive/folders/1IpGAPbiHYDJFsTtlBEktLGq6niTvB_BJ?usp=sharing).
> 
> Please let me know if you can access it so that I can close this issue.

**MY IDEA**
[image]

kennethjiang, 2020-11-17:
There are many different image labeling tools you can use. We use https://github.com/puzzledqs/BBox-Label-Tool internally. It's probably not the best but it's free. :)

CoderChen01, 2020-11-18:
> There are many different image labeling tools you can use. We use https://github.com/puzzledqs/BBox-Label-Tool internally. It's probably not the best but it's free. :)

Ok thanks for your reply, I use [https://github.com/tzutalin/labelImg](https://github.com/tzutalin/labelImg) , but you seem to have misunderstood my question and my question is how do you label your dataset to make the model work best like multiple target boxes on the spaghetti or on big target box on the whole spaghetti. as follows

**This is my old work**
[image]

**This is my new idea**
[image]

**Please guide me. Thank you very much**

hbrylkowski, 2020-11-18:
Thanks @kennethjiang got it, are there labels, or if I would want to try my approach I should label them myself?

kennethjiang, 2020-11-18:
Good question. Our way of labelling the images has evolved, from the one closer to your first approach to the ones close to to your 2nd lol. But it's been hard for us to go back to relabel images every time we adjust our way, so there is some (significant) inconsistency in our labels.

CoderChen01, 2020-11-19:
> Good question. Our way of labelling the images has evolved, from the one closer to your first approach to the ones close to to your 2nd lol. But it's been hard for us to go back to relabel images every time we adjust our way, so there is some (significant) inconsistency in our labels.

Thanks for you beyond words! I think the second method is better than the first one. What do you think?

kennethjiang, 2020-11-19:
That's what our intuition told us too. Interestingly enough, we didn't see as significant improvement as we hoped after switching to the new way of labelling the images. We are still not sure the shortfall in accuracy improvement was because Yolo is not very sensitive to how the data is labelled, or due to the fact that our training data is a mix of both labelling approaches (actually more as we evolved it).

CoderChen01, 2020-11-19:
> That's what our intuition told us too. Interestingly enough, we didn't see as significant improvement as we hoped after switching to the new way of labelling the images. We are still not sure the [...]
OK, thank you. Thank you so much. Best wishes to `Thespaghetdetective`

closed 2022-04-26 by raymondh2