TheSpaghettiDetective/obico-server #327: Discord notifications fail on Jetson Nano

amolloy, 2020-12-06:
I recently tried setting up Discord notifications on my local TSD install on a Jetson Nano. Unfortunately, I have yet to receive any notification. Grepping the logs for "discord" reveals complaints about SSL certificate verification failing:

```
-> % sudo docker-compose logs | grep -i discord
tasks_1   | [2020-12-03 02:00:29,753: ERROR/ForkPoolWorker-2] HTTPSConnectionPool(host='discord.com', port=443): Max retries exceeded with url: /api/webhooks/-----------------/-------------------------------------------------------------------l (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1091)')))
```

(webhook URL redacted)

After googling around a bit, I did attempt adding this to web/Dockerfile:

```
RUN mkdir /etc/pki
RUN mkdir /etc/pki/tls
RUN mkdir /etc/pki/tls/certs
RUN apt-get install wget
RUN wget http://curl.haxx.se/ca/cacert.pem
RUN mv cacert.pem /etc/pki/tls/certs/ca-bundle.crt
```

and rebuilt, but that made no difference.

Any ideas? Thanks!

Side note - it would be nice if all of the various notifications methods had a "test" button in the preferences UI. Thanks again!

kennethjiang, 2020-12-06:
I'm not familiar with how discord webhook works.

@chand1012 Any idea?

chand1012, 2020-12-06:
Are you able to execute the hook using `curl` on both the nano and a normal machine? 

`curl -X POST -H "Content-Type: application/json" -d '{"username": "test", "content": "hello"}' "<webhook url>"`

amolloy, 2020-12-07:
Weirdly, the first time I tried that command on the nano, it produced no output and had an error code of 0. But, a few minutes later, I tried it again and got the following:

```
curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: https://curl.haxx.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.
```

On my mac, curl always produces no output and returns an error code of 0.

Incidentally, I had a print complete in the middle of the night, and the notification appeared on my discord server as expected. But, another print completed just a few minutes ago and no notification appeared.

I'm tempted to just reinstall the OS on the nano. TSD is currently the only thing running on it, so it wouldn't be a big deal to get it set up again.

chand1012, 2020-12-07:
@amolloy That's what I would do as well, although I'm tempted not to call this an OS issue since TSD is running on Docker and effectively has a separate OS, but I find it very curious that the issue persists between both the host OS and TSD.

amolloy, 2020-12-08:
Well, reinstalling the OS did not resolve the issue... But I did figure it out, and it turns out to be user error! Sorry for taking your time on this, turned out to be a problem with my local network configuration. In particular, my router has some parental controls, and the Nano had gotten placed in its default profile which (you guessed it!) filters out Discord. I figured it out when I noticed that 
```
nslookup discord.com
```
returned completely different results on my Mac and on the Nano, and the IP returned on the Nano was a private (10.x.x.x) address.
Anyhow, thank you for helping diagnose this, and sorry again for wasting your time! I appreciate it. :)

closed 2020-12-08 by amolloy