TheSpaghettiDetective/obico-server #328: Bad handshake from traefik

elightcap, 2020-12-10:
Im using a pretty default traefik config and am getting some errors. traefik is showing level=error msg="vulcand/oxy/forward/websocket: Error dialing \"thespaghettidetective.(my domain is here)\": websocket: bad handshake with resp: 500 500 Internal server error"

while the actual TSDweb container is showing 

`django.channels.server INFO     WebSocket DISCONNECT /ws/dev/ [172.31.0.2:53946]
django.channels.server INFO     WebSocket HANDSHAKING /ws/dev/ [172.31.0.2:53950]
daphne.server ERROR    Exception inside application: [Errno -2] Name does not resolve
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/channels/utils.py", line 51, in await_many_dispatch
    await dispatch(result)
  File "/usr/local/lib/python3.7/site-packages/asgiref/sync.py", line 296, in __call__
    ret = await asyncio.wait_for(future, timeout=None)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 414, in wait_for
    return await fut
  File "/usr/local/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.7/site-packages/channels/db.py", line 14, in thread_handler
    return super().thread_handler(loop, *args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/asgiref/sync.py", line 334, in thread_handler
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/channels/consumer.py", line 105, in dispatch
    handler(message)
  File "/usr/local/lib/python3.7/site-packages/channels/generic/websocket.py", line 39, in websocket_connect
    self.connect()
  File "/usr/local/lib/python3.7/site-packages/newrelic/api/background_task.py", line 103, in wrapper
    return wrapped(*args, **kwargs)
  File "/app/api/consumers.py", line 82, in connect
    self.channel_name
  File "/usr/local/lib/python3.7/site-packages/asgiref/sync.py", line 139, in __call__
    return call_result.result()
  File "/usr/local/lib/python3.7/concurrent/futures/_base.py", line 428, in result
    return self.__get_result()
  File "/usr/local/lib/python3.7/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
  File "/usr/local/lib/python3.7/site-packages/asgiref/sync.py", line 204, in main_wrap
    result = await self.awaitable(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/channels_redis/core.py", line 602, in group_add
    async with self.connection(self.consistent_hash(group)) as connection:
  File "/usr/local/lib/python3.7/site-packages/channels_redis/core.py", line 864, in __aenter__
    self.conn = await self.pool.pop()
  File "/usr/local/lib/python3.7/site-packages/channels_redis/core.py", line 79, in pop
    conn = await aioredis.create_redis(**self.host, loop=loop)
  File "/usr/local/lib/python3.7/site-packages/aioredis/commands/__init__.py", line 175, in create_redis
    loop=loop)
  File "/usr/local/lib/python3.7/site-packages/aioredis/connection.py", line 113, in create_connection
    timeout)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 414, in wait_for
    return await fut
  File "/usr/local/lib/python3.7/site-packages/aioredis/stream.py", line 24, in open_connection
    lambda: protocol, host, port, **kwds)
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 913, in create_connection
    type=socket.SOCK_STREAM, proto=proto, flags=flags, loop=self)
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 1290, in _ensure_resolved
    proto=proto, flags=flags)
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 792, in getaddrinfo
    None, getaddr_func, host, port, family, type, proto, flags)
  File "/usr/local/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.7/socket.py", line 752, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -2] Name does not resolve

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/channels/sessions.py", line 183, in __call__
    return await self.inner(receive, self.send)
  File "/usr/local/lib/python3.7/site-packages/channels/middleware.py", line 41, in coroutine_call
    await inner_instance(receive, send)
  File "/usr/local/lib/python3.7/site-packages/channels/consumer.py", line 59, in __call__
    [receive, self.channel_receive], self.dispatch
  File "/usr/local/lib/python3.7/site-packages/channels/utils.py", line 58, in await_many_dispatch
    await task
  File "/usr/local/lib/python3.7/site-packages/channels_redis/core.py", line 453, in receive
    real_channel
  File "/usr/local/lib/python3.7/site-packages/channels_redis/core.py", line 508, in receive_single
    index, channel_key, timeout=self.brpop_timeout
  File "/usr/local/lib/python3.7/site-packages/channels_redis/core.py", line 340, in _brpop_with_clean
    async with self.connection(index) as connection:
  File "/usr/local/lib/python3.7/site-packages/channels_redis/core.py", line 864, in __aenter__
    self.conn = await self.pool.pop()
  File "/usr/local/lib/python3.7/site-packages/channels_redis/core.py", line 79, in pop
    conn = await aioredis.create_redis(**self.host, loop=loop)
  File "/usr/local/lib/python3.7/site-packages/aioredis/commands/__init__.py", line 175, in create_redis
    loop=loop)
  File "/usr/local/lib/python3.7/site-packages/aioredis/connection.py", line 113, in create_connection
    timeout)
  File "/usr/local/lib/python3.7/asyncio/tasks.py", line 414, in wait_for
    return await fut
  File "/usr/local/lib/python3.7/site-packages/aioredis/stream.py", line 24, in open_connection
    lambda: protocol, host, port, **kwds)
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 913, in create_connection
    type=socket.SOCK_STREAM, proto=proto, flags=flags, loop=self)
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 1290, in _ensure_resolved
    proto=proto, flags=flags)
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 792, in getaddrinfo
    None, getaddr_func, host, port, family, type, proto, flags)
  File "/usr/local/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.7/socket.py", line 752, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -2] Name does not resolve
django.channels.server INFO     WebSocket DISCONNECT /ws/dev/ [172.31.0.2:53950]`

The web frontend actually works fine for me, but octopi is unable to connect to it, and im not really sure where else to look at this point.  I do see traefik trying to get ssl certs for some weird domains that i definitely do not have cnames for, but not sure that has anything to do with it.

`level=error msg="Unable to obtain ACME certificate for domains \"strange-hawking.mydomaincom\" detected thanks to rule \"Host:strange-hawking.mydomain.com\" : unable to generate a certificate for the domains [strange-hawking.mydomain.com]: acme: Error -> One or more domains had a problem:\n[strange-hawking.mydomain.com] acme: Error 400 - urn:ietf:params:acme:error:dns - DNS problem: NXDOMAIN looking up A for strange-hawking.mydomain.com - check that a DNS record exists for this domain\n"
time="2020-12-10T02:07:44Z" level=error msg="Unable to obtain ACME certificate for domains \"exciting-ramanujan.mydomain.com\"`

Thanks for any help you can provide!

elightcap, 2020-12-11:
checking traefik, it looks like octopi is trying to request to myhostname.com/ws/dev but i dont see that existing anywhere in the web dir

kennethjiang, 2020-12-11:
What did you set in the "Server Address" field in the TSD plugin? https://github.com/TheSpaghettiDetective/TheSpaghettiDetective#configure-the-spaghetti-detective-octoprint-plugin-to-use-your-own-server

elightcap, 2020-12-11:
https://thespaghettidetective.mydomain.com.

Seems to be an issue with putting it behind traefik.  It works fine when i dont reverse proxy it, over port 3334.  Im currently using traefik 2.3.5, but id di also try it with 1.7 and got the same result.  Python isnt my first language, but it looks like that URL is suppose to come from /web/api/ws_routing.py, but traefik just doesnt seem to be routing there

kennethjiang, 2020-12-11:
One possible cause is the plugin needs to connect to websocket too. in this case it'll be `wss://thespaghettidetective.mydomain.com`. You want to make sure traefik is properly configured to handle that.

elightcap, 2020-12-11:
interesting ill have to look into that a bit more, not sure how traefik would handle it

elightcap, 2020-12-13:
well i was able to hack a way around this.  I ran the docker-compose on its own, so port 3334 was exposed to the internet.  Then built a dynamic config routing 443 traffic to `tsd.mydomain.com` to the docker container.  Works now!

kennethjiang, 2020-12-13:
I'm closing this issue as this is not a code issue, and a workaround has been found. The closed issue will still be searchable by google for whoever later run into the same issue

closed 2020-12-13 by kennethjiang